Xinyuan-LilyGO/LilyGo-T-SIM7080G #3: ATT NBIOT Card Registration Issue

wenjun1972, 2023-03-29:
Hi, I inserted an ATT NBIOT SIM card on 7080G and program the example code MinimalModemNBIOTExample through Platformio, whereas, the console port I saw continuously repeat the messages as below:

OK
AT+CGREG?

+CGREG: 0,2

OK
.AT+CEREG?

+CEREG: 0,2

OK
AT+CGREG?

+CGREG: 0,2
........

These messages keep repeating and won't stop, is it normal ? The LED close to CHD keeps blink as well, what does it mean?

In addition, according to ATT NBIOT doc, ATT NB-IoT devices should be setup to use APN "m2mNB16.com.attz", where shall I put it inside the code, line 53 in the code says the APN there is for GPRS.

lewisxhe, 2023-03-29:
+CGREG: 0,2 This is an action that can only be executed after the APN setting has been registered while searching for signals and waiting for registration. Please refer to the manual for specific command meanings

lewisxhe, 2023-03-29:
https://github.com/Xinyuan-LilyGO/LilyGo-T-SIM7080G/commit/9ea916e3412e8faa9db61fa6a49753919dec61dc

wenjun1972, 2023-03-29:
it changed to +CGREG: 0,3 now after programed your updated code, per the AT command spec, "3" means Registration denied, The GPRS service is disabled, the UE is not allowed to attach for GPRS if it is requested by the user.

wenjun1972, 2023-03-29:
it works well now after registration this SIM in ATT account, thanks !

closed 2023-03-29 by wenjun1972