Xinyuan-LilyGO/LilyGo-T-SIM7080G #4: AWS MQTT Access

wenjun1972, 2023-03-29:
Hi, do you have examples to access AWS MQTT through the RootCA, device cert and private key ?

wenjun1972, 2023-03-31:
Hi, do you have any successful code to connect AWS MQTT? My try was failed at the last step at AT+SMSSL as below.

AT+CSSLCFG="CONVERT",2,"rootCA.crt"

OK
AT+CSSLCFG="CONVERT",1,"deviceCert.crt","devicePrivateKey.pem"

OK
AT+SMSSL=1,"rootCA.crt","deviceCert.crt"

OK
AT+SMCONN

ERROR

lewisxhe, 2023-03-31:
I uploaded a certificate writing example, but it seems that my network problem has caused me to be unable to connect. Could you help me test whether I can connect normally?

https://github.com/Xinyuan-LilyGO/LilyGo-T-SIM7080G/blob/master/examples/ModemMqttsAuthExample/ModemMqttsAuthExample.ino

lewisxhe, 2023-04-01:
Thank you for your test. I'm a bit skeptical that I can't log in through the 8884 encrypted port. I'll try using the 1883 encrypted port for the certificate login test. If there's a message, I'll update it

wenjun1972, 2023-04-01:
Got it, do you have successful case of other T-SIM product connected to AWS through SSL certificates ?

wenjun1972, 2023-04-01:
The T-SIM7080G firmware is Revision:1951B05SIM7080, is there a way to upgrade to the latest version to try this?

lewisxhe, 2023-04-03:
I have previously tried to use encryption to log in to MQTTS in the SIM7600 and A7608 modules, using the same method. Oh! By the way, SIM7080 and SIM7000 have the same architecture. I have been trying to use encrypted login on SIM7000, but it has always failed. I think this may be the reason. Regarding the latest firmware, I have a B12 firmware, and you can try it out. Recently, I have too many things to do, and I will postpone testing this later
This is the update firmware address https://1drv.ms/u/s!AmbpOqVezk5dsVv9WzIaCx8eZGit?e=XjhxgC

wenjun1972, 2023-04-03:
Please provide the firmware upgrade tools, thanks.

lewisxhe, 2023-04-03:
https://github.com/Xinyuan-LilyGO/LilyGo-T-SIM7080G/blob/master/docs/sim7080_update_firmware.md

wenjun1972, 2023-04-03:
Unfortunately T-SIM7080G does not show up in my windows 11, it seems only working on windows 10.

https://github.com/Xinyuan-LilyGO/LilyGo-T-SIM7080G/blob/master/docs/sim7080_update_firmware.md

lewisxhe, 2023-04-03:
Did you burn [MinimalModemUpgrade](https://github.com/Xinyuan-LilyGO/LilyGo-T-SIM7080G/tree/master/examples/MinimalModemUpgrade ), and then press and hold the SBOOT button on the SIM7080G, and then insert a micro USB into the computer?"? Win11 and Win10 are the same

wenjun1972, 2023-04-03:
Had a tried but didn’t work (nothing happened) at the last step, will try to find a win10 to run the upgrade.

Understood you are busy as of now, but please make your time to explore this issue as critical since AWS access is very popular in global.

Thanks,

Jun

lewisxhe, 2023-04-04:
Let me clarify the upgrade process again
1. Burn MinimalModemUpgrade, disconnect USB after burning is complete
2. Press and hold the SBOOT button on the board (the first button next to MicroUSB), and insert MicroUSB
3. Keep the SBOOT button pressed, and insert the USB. At this time, you can see the QDloader 9008 (if the driver is installed) in the Device Manager. If the driver is not installed, you need to install the driver

lewisxhe, 2023-04-04:
AWS is in an inaccessible state in my country, and I will try to connect to other IoT platforms during my free time

wenjun1972, 2023-04-04:
I tried your upgrade process again on my windows 11 desktop and a windows 11 virtual machine on mac, even I installed QDloader 9008 driver,  but both didn't show up "QDloader 9008" in windows 11 device manager, but only reported additional COM3 port. I also tried another new T-SIM7080G board but saw the same issues.

wenjun1972, 2023-04-06:
Hi, I well upgraded to B12 firmware today but still had the issue of AT+SMCONN error.

lewisxhe, 2023-04-06:
I can guess it shouldn't be a version issue, and I am currently unable to determine the reason for this issue
What's annoying is that even if I can't connect, I can't see the cause of the error. I need to seek support from SIMCOM and hope they can respond positively to my question
Also, you can try taking a look at the following two libraries to see if they can be helpful to you. Currently, I have too many things to try. If there are good results, please let me know

https://github.com/OPEnSLab-OSU/SSLClient
https://github.com/govorox/SSLClient

wenjun1972, 2023-04-06:
I am highly doubtful if ATT NB-IOT network block the SSL port.

https://github.com/OPEnSLab-OSU/SSLClient
https://github.com/govorox/SSLClient

wenjun1972, 2023-04-06:
Had a try on "test.mosquitto.org" by my own certs and key, and saw the same error. If SSL port 8884 is blocked by ATT, how does SIM7000 and 7600 work in ATT NB-IOT with AWS and Mosquitto ?.

13:32:32:357 -> AT+CSSLCFG="CONVERT",2,"rootCA.pem"
13:32:32:458 -> 
13:32:32:458 -> OK
13:32:32:458 -> AT+CSSLCFG="CONVERT",1,"deviceCert.crt","devicePrivateKey.pem"
13:32:32:559 -> 
13:32:32:559 -> OK
13:32:32:559 -> AT+SMSSL=1,"rootCA.pem","deviceCert.crt"
13:32:32:565 -> 
13:32:32:565 -> OK
13:32:32:566 -> Connecting to test.mosquitto.org ...
13:32:32:566 -> AT+SMCONN
13:33:32:567 -> No valid response, retrying connect ...
13:33:33:568 -> AT+SMCONN
13:34:33:569 -> No valid response, retrying connect ...
13:34:34:569 -> AT+SMCONN
13:34:36:808 -> 
13:34:36:808 -> ERROR
13:34:36:808 -> Connect failed

lewisxhe, 2023-04-07:
SIM7000, you don't need to try it, because I have tried it before. I also failed.

It is no problem to connect to SIM7600 before the manual, but I use 4G. I don't know if the NB network will also block port 433.

wenjun1972, 2023-04-07:
A good news ! Just now I can access test.mosquitto.org through 8884 after correctly configure the AT&T APN  by the code - modem.sendAT("+CGDCONT=1,\"IP\",\"m2mNB16.com.attz\"");

+SMSTATE: 1

OK

lewisxhe, 2023-04-07:
Really ? Just set up APN?

wenjun1972, 2023-04-07:
It seems yes, but will try to access AWS today, at least it shows 8884 is not blocked by ATT.

lewisxhe, 2023-04-07:
Really good news, please continue to try. If you can, please submit a PR connecting AWS here. Thank you

wenjun1972, 2023-04-07:
Sure, thank you !

wenjun1972, 2023-04-10:
Hi, SSL to AWS IOT Core and http/https to www.mockable.io were tested successfully this weekend, I will submit a PR here soon, thanks for your support, we can close this issue now.

closed 2023-04-10 by wenjun1972

wenjun1972, 2023-04-13:
Hi, Lewis,

I posted the sample codes in https://github.com/bootcampiot, please let me know how to submit a PR in your example’s repository.

Thanks

Jun