Xinyuan-LilyGO/LilyGo-T-SIM7080G #8: MQTT Publish Issues

bootcampiot, 2023-04-12:
Hi,

I found ModemMqttPulishExample the snprintf format below was not aligned with the parameters definition of SMPUB in SIMCOM MQTT application documentation, username, clientID and data_channel are not required here.
---------------------------------------------------------------------------------------------------------------
    // AT+SMPUB=<topic>,<content length>,<qos>,<retain><CR>message is enteredQuit edit mode if messagel ength equals to <contentlength>
    snprintf(buffer, 1024, "+SMPUB=\"v1/%s/things/%s/data/%d\",%d,1,1", username, clientID, data_channel, payload.length());
 ---------------------------------------------------------------------------------------------------------------
I am trying to construct below SMPUB to publish topic "device_connect" to AWS IOT Core, but saw "ERROR" response and no data arrived at AWS IOT Core.

snprintf(buffer, 1024, "+SMPUB=\"%s\",%d,1,1", pub_topic.c_str(), payload_rtcMem.length());
(output is AT+SMPUB="$aws/things/xxxxxxxxx/event_type/device_connect",339,1,1)
modem.sendAT(buffer);
modem.stream.write(payload_rtcMem.c_str(), payload_rtcMem.length());

>{payload}
 
ERROR

wenjun1972, 2023-04-13:
Hi, Lewis,

Please advise about this issue.

Thanks,

Jun

lewisxhe, 2023-04-14:
Sorry, I didn't notice , I'll reply later

wenjun1972, 2023-04-14:
The missed "\\" was the issue from github comment writing, it was stripped out in my issue report text, very strange here. Both of my publish and subscribe didn't work, no message arrive at AWS IOT Core, only the presence of device connect and disconnect.

My publish shows as below
>snprintf(buffer, 1024, "+SMPUB=\\"%s\\",%d,1,1", pub_topic.c_str(), payload_rtcMem.length());

My subscribe shows as below
>snprintf(buffer, 1024, "+SMSUB=\\"%s\\",1", sub_topic.c_str())

lewisxhe, 2023-04-18:
Subscribe and publish commands are valid if you fill in the parameters according to the manual, at least I have tried it, and you can use a free public MQTT server to test
https://www.emqx.com/zh/mqtt/public-mqtt5-broker

wenjun1972, 2023-04-18:
I had a try over the weekend, this pub and sub format worked well on cayance.mydevice.com, test.mosquitto.com, but never success at AWS IOT Core, I guess this format only works on the MQTT basic authentication approach by username, password and clientID.

lewisxhe, 2023-05-22:
Any progress?

wenjun1972, 2023-05-22:
We tried with huge effort but failed to make MQTT publish working with AWS, now we change to other SIM platform and applied UDP approach which is good to our use case. Did you have any test done for MQTT publish with AWS ?

lewisxhe, 2023-05-23:
Sorry, I haven't completed it yet. I am currently contacting technical support from Simcom. I will update you promptly if there is any news

junwenatsmartguard, 2023-05-25:
Hi, Lewis,

We made good progress today, MQTT published messages were well received at AWS core.

The changes we made were to change qos from 1 to 0 and retain from 1 to 0 in ----> AT+SMPUB=\"%s\",%d,0,0", pub_topic.c_str(), pub_payload.length()

We can close this issue now.

lewisxhe, 2023-05-27:
Good news! If the test is successful, please submit a PR.

bootcampiot, 2023-05-30:
Will submit PR this week.

closed 2023-05-30 by bootcampiot

wenjun1972, 2023-05-31:
Hi, Lewis,

I just submitted 3 PRs, please check.

Regards,

Jun