Xinyuan-LilyGO/LilyGo-T-SIM7080G #9: ESP32 does not come out of deep_sleep when running on battery

gfilippi, 2023-04-26:
I'm testing the LilyGo-T-SIM7080 for deep_sleep mode and it seems that there is an issue.

I'm using a simple loop

void loop()
{ 
    esp_sleep_enable_timer_wakeup(5 * ONESEC_USECS);
    LOG_MSGLN("Entering deep sleep\n");
    esp_deep_sleep_start();
}

1) if I power the device using the USB connector I can see the device going in deep_sleep and rebooting every 5 secs
2) if I power the device from battery I can see the first boot, then it goes in deep sleep ... but it will never come back.

it seems to me that there is some issue with the PMU not providing 3.3v when going in deep sleep and running from battery.

note: my LiPo battery is fully charged at 4.1v

lewisxhe, 2023-04-27:
The 7080G board's scheduled deep sleep wake-up is no problem at all. I have tested it. [Example](https://github.com/Xinyuan-LilyGO/LilyGo-T-SIM7080G/blob/9f63ab0ae8a5af524ca44d59d443ccac0469e4d5/examples/BIGIOT_Gnss_Upload/BIGIOT_Gnss_Upload. ino Timed wakeup is also used in #L110), it has been running for a while, it works fine, you can write this sketch for a simple test.
https://github.com/espressif/arduino-esp32/blob/master/libraries/ESP32/examples/DeepSleep/TimerWakeUp/TimerWakeUp.ino

Also, please remember that if you want to connect to the computer without connecting the USB, you must turn off the USB CDC enable, otherwise every time you start it will block and wait for the USB to be connected. This is in the ini It has been stated in the document, it seems that your problem is more like not turning off the USB CDC
https://github.com/Xinyuan-LilyGO/LilyGo-T-SIM7080G/blob/9f63ab0ae8a5af524ca44d59d443ccac0469e4d5/README.MD?plain=1#L54

gfilippi, 2023-04-28:
USB CDC is disabled in my setup. I confirm the problem.

to exploit you just need to connect an ammeter between the battery and the board to measure the "deep sleep current". The moment you put the system in deep sleep .. PMU will turn down power.

I have a workaround by interconnecting a "uCurrent by EEVBlog" instrument instead of an ammeter (btw: I tested multiple ammeters ...). If I use the dedicated "uCurrent" instrument everything works as expected and I can measure currents in all the different states of powersave/sleep cycling every 5s using the timer wakeup

lewisxhe, 2023-04-28:
So I know what the reason is because you connected an ammeter in series between the PMU and the battery, causing the starting current to decrease and causing the PMU to believe that the power supply capacity is insufficient, so the PMU will not start

gfilippi, 2023-04-28:
ok, understood. So how normally do you measure the power consumption on your battery? ... I mean .. I have a solution withthe uCurrent/EEVBLog but I'm wondering how you test it in your lab. Thanks a lot for your help

lewisxhe, 2023-04-28:
We are using a digital ammeter. Here is a picture
https://github.com/Xinyuan-LilyGO/LilyGo-T-SIM7080G/tree/master/examples/BIGIOT_Gnss_Upload

gfilippi, 2023-04-28:
very nice! could you please send a picture with connections? it is not clear where you apply power and where is the ammeter inserted in series. Thanks a lot

lewisxhe, 2023-05-04:
This is a schematic diagram
[image]

lewisxhe, 2023-05-22:
If you still have problems, please reopen

closed 2023-05-22 by lewisxhe