Xinyuan-LilyGO/LilyGo-T-SIM7080G #10: where to get IIC pin

gfilippi, 2023-04-29:
I would like to connect an external I2C device but it is not clear to me where we can get the same IIC_SDA IIC_SCLK that is used for the PMU. Which pin should we use for electrical connection to the system I2C bus?

lewisxhe, 2023-04-30:
PMU SDA SCL is not led out. Please use another IO port as the interface of the IIC device. You can use IO1 (SDA) and IO2 (SCL) as the access of external IIC devices. By default, pull-up resistor has been added to these two IOs

lewisxhe, 2023-05-04:
If there are still issues, please reopen it

closed 2023-05-04 by lewisxhe

vabatta, 2023-06-19:
Out of curiosity, which was the reason to not pull out the SDA SCL pin mapping connected to the PMU? This would have left more devices being able to make use of one of the 2 I2C controllers of the MCU.