Xinyuan-LilyGO/LilyGo-T-SIM7080G #11: schematic and homepage GPIO mismatch

gfilippi, 2023-04-29:
Hello, I'm trying to use the GPIO46 and I believe that there is a mismatch between the schematic and the illustration of the board from the home page: https://www.lilygo.cc/products/t-sim7080-s3

could you please clarify which one is correct? it seems to me that for example pin 9 is not matching what is reported on the schematic.

lewisxhe, 2023-04-30:
oh There is an error in the pin diagram. Thank you for your feedback. Please refer to the board schematic and the silk screen on the board for details
[image]

gfilippi, 2023-04-30:
I verified each pin manually and it seems to me that the silkscreen is correct, while the schematic is wrong.
Could you please upload the correct schematic that matches the current board?

thanks

lewisxhe, 2023-04-30:
[image]
The schematic is not incorrect, and I am certain it is correct

gfilippi, 2023-04-30:
Hi lewis, on the schematic P2 is exposing CAM_D2 on pin 3 and CAM_D6 on pin 4, but D2 is IO14 and D6 is IO13.

instead on P2 we have IO17 and IO18.

please confirm

lewisxhe, 2023-04-30:
[image]

I understand what you mean, but the starting position of the needle arrangement is different, and the understanding is different.

lewisxhe, 2023-04-30:
In order to reduce misunderstandings, I have reversed the Pin header

closed 2023-04-30 by lewisxhe