Xinyuan-LilyGO/LilyGo-T-SIM7080G #12: PWRKEY PIN

wenjun1972, 2023-05-01:
Hi,

What is the PWRKEY pin ? 7080G pinout tells GPIO41 for PWR, is it correct one? 

I expect to set the PSM mode and wake up the modem by pulling LOW from HIGH, but I read the status of that PIN is always LOW.

Thanks,

Jun

wenjun1972, 2023-05-03:
> PWRKEY is the power on function pin of PMU, which is not physically connected to ESP32

Can you point to me any example code to pull low this function pin to exit from PSM status ?

lewisxhe, 2023-05-03:
Sorry, I misunderstood, PWRKEY exists and it is connected to IO41, you wake up from PSM mode, you need to pull down PWRKEY, there is a level conversion circuit in the hardware, so you need to set IO41 to HIGH, this When PWRKEY will be LOW

lewisxhe, 2023-05-22:
If you still have problems, please reopen

closed 2023-05-22 by lewisxhe

ibethcompany, 2023-05-30:
Hello, I'm using the battery and when the battery runs out or I remove the battery from the board and return it, the program does not start, I have to press the physical PWR button on the board for it to work...

Is there any way to make it so that when the battery dies or I remove it and put it back in, the program will work normally again without having to press the PWR button on the board?

Thank you very much @lewisxhe

lewisxhe, 2023-05-31:
If you need to connect the battery to automatically power on, it cannot be done on the current hardware

Another way is to modify the hardware. You need to connect the PMU_IRQ and GND in the picture below. At this time, the battery will be connected to the computer and it will automatically start up. However, the disadvantage is that the interrupt signal of the PMU cannot be used.
[image]