Xinyuan-LilyGO/LilyGo-T-SIM7080G #13: LilyGo-T-SIM7080G power consumption in deep sleep

robizzar, 2023-05-05:
What about your measurement?
It is very high.

At switch-on (PMU is still off) the current drawn is already 240uA.
Pressing button PWRKEY to start my ESP32s3 firmware, which simply goes into deep sleep turning off all unused PMU outputs, the current draw increases to 1.2mA.
Thanks.

lewisxhe, 2023-05-05:
I have described the power consumption during sleep in the [README](https://github.com/Xinyuan-LilyGO/LilyGo-T-SIM7080G/blob/304a499a4cfef5fe65b671813c1606427214b95b/README.MD?plain=1#L17), which is also 1.2mA, and I am not looking for the reason why the power consumption is so high.

Other sleep consumption using PMU is about 400uA, SIM7080 currently I do not know what is causing it

robizzar, 2023-05-05:
In your test, in deep sleep your power consumption is 1.2mA as mine.
Is it right?

What do you mean for  _sleep consumption using PMU_?  I do not understand this..

robizzar, 2023-05-05:
The other versions of LILYGO (SIM7000G and A7670E) consume much less current.
Do you think you will try to find the cause of this problem?

4ib3r, 2023-05-17:
I had the same problem, in my case lowering DC1 voltage to 2500mV before esp32 enter deep sleep (and back to on wake-up 3300mV) allowed me to decrase current consumption from battery below 0.6mA.

lewisxhe, 2023-05-22:
Because of the use of the PMU, the deep sleep current consumption will increase, which cannot be solved on the current version of the hardware.

theMladyPan, 2023-06-22:
Hi all, Can you please provide minimal example for deep sleep? I have been tinkering around but cannot go below 25mA (Ive used snippets from the examples/BIGIOT_Gnss_Upload/BIGIOT_Gnss_Upload.ino but without success. The device goes to deep sleep (Serial is terminated) and but power draw remains at 25mA. No external components attached only battery (No USB either).

hugo8873, 2024-03-12:
Hi all, I might archived an even lower power consumption, but I dont have any gadget to measure it in uA level.  You guys want to have a try and share the results? @lewisxhe @theMladyPan @robizzar 

From last night 23:49, I measured a 4.132v voltage, and in this afternoon 13:53, It is 4.116v, in 14 hours, it only had 16mV drops.  Im using a small 230mAh Li-on batter. 

In my void anySleepFucntion(); I did following steps:
1. modem.poweroff();
2. while (modem.testAT()) {
      Serial.println("Modem still active...");
      delay(1000);
    }
3. Disable all ADC related functions by calling:
  PMU.disableTemperatureMeasure();
  PMU.disableBattDetection();
  PMU.disableVbusVoltageMeasure();
  PMU.disableBattVoltageMeasure();
  PMU.disableSystemVoltageMeasure();
4. Turn off all DCs, LODs and CPULDO
5. end Serials and wires,
6. and eventually, reset used GPIOs in my sketch by gpio_reset_pin(GPIO_NUM_?);

Check the codes;
```
void setupSleepMode(void) {
  PMU.setChargingLedMode(XPOWERS_CHG_LED_BLINK_4HZ);
  // Turn off modem
  modem.poweroff();
  // Wait until the modem does not respond to the command, and then proceed to the next step
  while (modem.testAT()) { //double while() check
    delay(1000);
    while (modem.testAT()) {
      Serial.println("Modem still active...");
      delay(1000);
    }
  }
  Serial.println("Modem not responding. Modem turned off.");

  // Turn off ADC data monitoring to save power
  PMU.disableTemperatureMeasure();
  PMU.disableBattDetection();
  PMU.disableVbusVoltageMeasure();
  PMU.disableBattVoltageMeasure();
  PMU.disableSystemVoltageMeasure();

  // Reserve the MCU chip power supply, LilyGo AXP2101 usually uses DC as ESP power supply
  //PMU.enableDC1();

  // Turn off the power output of other channels
  PMU.disableDC2();
  PMU.disableDC3();
  PMU.disableDC4();
  PMU.disableDC5();
  Serial.println("DC turned off");
  PMU.disableALDO1();
  PMU.disableALDO2();
  PMU.disableALDO3();
  PMU.disableALDO4();
  PMU.disableBLDO1();
  PMU.disableBLDO2();
  PMU.disableCPUSLDO();
  PMU.disableDLDO1();
  PMU.disableDLDO2();
  Serial.println("A,B,C,D,LDO turned off.");

  Serial1.end();
  Serial1.end(); // Not sure why end Serial1 twice, it was copied from example
  Serial.println("Serial1 end.");

  // Clear PMU Interrupt Status Register // I applied a short press interrupt on PMU
  PMU.clearIrqStatus();
  Serial.println("PMU IRQ status cleared.");

  // Turn off the charging indicator to save power
  PMU.setChargingLedMode(XPOWERS_CHG_LED_OFF);

  Wire.end();
  Wire1.end();
  Serial.println("Wire & wire1 end");

// reset pins I used in my sketch
  gpio_reset_pin(GPIO_NUM_4);
  gpio_reset_pin(GPIO_NUM_5);
  gpio_reset_pin(GPIO_NUM_7);
  gpio_reset_pin(GPIO_NUM_15);
  gpio_reset_pin(GPIO_NUM_38);
  gpio_reset_pin(GPIO_NUM_39);
  gpio_reset_pin(GPIO_NUM_40);
  gpio_reset_pin(GPIO_NUM_41);
  gpio_reset_pin(GPIO_NUM_45);
  gpio_reset_pin(GPIO_NUM_46);
  gpio_reset_pin(GPIO_NUM_47);
  gpio_reset_pin(GPIO_NUM_48);
  Serial.println("gpio_reset_pins.");

  Serial.flush(); // copied from an example
  Serial.end();

#if !CONFIG_IDF_TARGET_ESP32S3
  Serial.println("Please implement the MCU sleep method");
#else
  // Set ESP32 to wake up externally
  //esp_sleep_enable_ext0_wakeup((gpio_num_t)pmu_irq_pin, LOW);
  esp_sleep_enable_ext0_wakeup(GPIO_NUM_6, 0);  //1 = High, 0 = Low //PMU_INPUT_PIN (6)
  Serial.println("Last step, wakeup pin defined");
  // Enable ESP32 sleep
  esp_deep_sleep_start();
#endif
}
```

manzajaro, 2024-03-21:
Hi @hugo8873 

**Full execution**

[image]

**Function setupSleepMode execution**
[image]

**Sleep**
[image]

manzajaro, 2024-03-21:
What is the best way to sleep the device and then have the modem work? Can someone attach an example?

hugo8873, 2024-03-26:
Hi @manzajaro,
Thanks so much, happy to see the current dropped to 315uA.  I guess the essential benefit came from the reset GPIO pins.
(The 100mA while running is also shockingly large, lol)

And if you cannot restart the modem, I think the problem might lies in the PMU interrupt, I found that power supplies can behave unexpected with incontinuous MCU wire communication.  I attached a short press PMU interrupt, and it restart my peripherals normally. 
Moreover, that I can think about is, if you upload a new sketch into the EPS32, it will naturally restart the chip without properly run through the power down sequence, therefore, peripherals may not initialize normally. You can use ESP.restart() in some peripheral initiation steps, and the board will eventually gets running.

You can find interrupt code from example folder and I also attach my here:
```
void setFlag(void) {
  pmu_flag = true;
}

//in setup()---
  pinMode(PMU_INPUT_PIN, INPUT_PULLUP);  // the shared pin 6 of SoC and PMU.  While DC1 power supply remained, SoC can be wakeup by this pin.  And PMU can also be forced power off by long press this pin.
  attachInterrupt(PMU_INPUT_PIN, setFlag, FALLING);
  // Close other IRQs
  PMU.disableIRQ(XPOWERS_AXP2101_ALL_IRQ);
  // Clear all interrupt flags
  PMU.clearIrqStatus();
  // Enable the required interrupt function
  PMU.enableIRQ(XPOWERS_AXP2101_PKEY_SHORT_IRQ);  //POWER KEY
```
Hope this helps.

hugo8873, 2024-03-27:
Dear @manzajaro ,
Good to hear from you again, and I found comments from example codes written like these,
```
  //! Do not turn off BLDO1, which controls the 3.3V power supply for level conversion.
  //! If it is turned off, it will not be able to communicate with the modem normally
  PMU.setBLDO1Voltage(3300);  //Set the power supply for level conversion to 3300mV
  PMU.enableBLDO1();
```
I enabled BLOD1 before DC3(modem), looks fine so far.

closed 2024-05-11 by github-actions[bot]

gtom84, 2025-01-08:
For anyone looking to further decrease deep sleep consumption, all of the above is great and in addition

from https://github.com/lewisxhe/XPowersLib/blob/master/examples/AXP2101_Sleep_Example/AXP2101_Sleep_Example.ino

> In the experiment on T-SIM7080-ESP32, the power consumption is about 900uA after setting the PMU to sleep, and 1mA if it is not set to sleep. After setting the PMU to sleep, it can save about 100uA consumption current

```
// Enable PMU sleep
power.enableSleep();

// Send IRQ wakeup command
power.enableWakeup();

```

KatmanCode, 2025-07-15:
@manzajaro Hi Friend,

Thanks for your suggestion can make my code down to 0.38mA, but I have observed the standby current could raise from 0.38mA to 1.8mA, a details description has been opened in issue #152, i would like to know if you have observed similar raise of current, or only my hardware or software problem.

Thanks for everything