Xinyuan-LilyGO/LilyGo-T-SIM7080G #15: Can not connect to serial monitor in arduino

wlueger, 2023-05-09:
Hi,

While i can upload a sketch, by holding boot btn, i can not get serial output:
Port monitor error: command 'open' failed: Invalid serial port. Could not connect to COM5 serial port.
on Windows 10!
Arduino IDE: 2.1.0
[image]

Any help would be appreciated!

Thanks, W.

lewisxhe, 2023-05-09:
Please refer to Article 5 in README's TIPS

wlueger, 2023-05-10:
> Please refer to Article 5 in README's TIPS

@lewisxhe Thank you very much!

Indeed, i set the proper pins for the modem (4,5,41) and get output now!
It seems that the system crashes before any output can be serialised...
This is strange behaviour compared to other boards, it seems that the serial board is handled differently.

Anyway i can proceed now - THX!

W.

closed 2023-05-22 by lewisxhe