Xinyuan-LilyGO/LilyGo-T-SIM7080G #17: Twilio NB-IOT sim card not registering

dustymac, 2023-05-20:
I have made sure to activate this sim card within my Twilio account and so I've made sure all Twilio steps are finished.
I have changed the APN to iot.nb according to Twilio's docs.

`   const char *apn = "iot.nb";   `

My console initalizes everything properly, but my CGREG and CEREG start out 0,2 for a few tries.  Then both are 0,0 for as long as I allow it to run.  

This is my first SIM card project.  Where do I need to start troubleshooting?

**CONSOLE:**

Modem started!AT+CPIN?
AT+CPIN?
+CPIN: READY

OK
AT+CNMP=2
AT+CNMP=2
OK
AT+CMNB=2
AT+CMNB=2
OK
AT+CMNB?
AT+CMNB?
+CMNB: 2

OK
AT+CNMP?
AT+CNMP?
+CNMP: 2

OK
getNetworkMode:2 getPreferredMode:2
AT+CEREG?
AT+CEREG?
+CEREG: 0,2

OK
AT+CGREG?
AT+CGREG?
+CGREG: 0,2

OK
.AT+CEREG?

SMS Ready
AT
AT+CEREG?
+CEREG: 0,2

OK
ATE0
AT
OK
AT+CMEE=0
ATE0
OK
AT+CLTS=1

OK
AT+CBATCHK=1

OK
AT+CPIN?

OK

+CPIN: READY

OK
AT+CGREG?

+CGREG: 0,2

OK
.AT+CEREG?

+CEREG: 0,0

OK
AT+CGREG?

+CGREG: 0,0

OK
.AT+CEREG?

+CEREG: 0,0

^^^^ LAST MESSAGE REPEATS FOREVER

lewisxhe, 2023-05-22:
You need to confirm whether your SIM card operator supports NB-IOT

dustymac, 2023-05-22:
T-Mobile shows complete coverage for my city. But after talking to other devs in the area they tell me the coverage map must be wrong because they have never been able to T-Mobile here.

Ordering a new SIM to use other carriers here. Thanks.

dustymac, 2023-05-26:
I have purchased a SIM card from Hologram which allows me to try other cell carriers in my town.  I am still having issues.  Cannot connect successfully, but am receiving:

.AT+CEREG?
+CEREG: 0,3

AT+CGREG?
+CGREG: 0,3

Please help me simplify the excellent example projects you have built and get down to the nuts and bolts of my connection.  Perhaps a basic example could be:  ConnectToLocalCarrierExample

Your help would be greatly appreciated.  Also, when I google around the internet I am finding lots of other users having similar issues with the 7080G module.  Just wanted to mention that.

dustymac, 2023-05-26:
Finally figured out the problem, check out my new issue titled "Using 7080G with Hologram"

lewisxhe, 2023-05-27:
If you still have problems, please reopen

closed 2023-05-27 by lewisxhe