Xinyuan-LilyGO/LilyGo-T-SIM7080G #18: Wake up esp from modem after exiting PSM

4ib3r, 2023-05-22:
Hello, I try to use PSM mode and wake-up esp using RI pin (gpio 03 in esp). Unfortunately signal from a modem come randomly. Modem wakes up from PSM and status led turn on regulary. 
How to properly wake up the esp from the modem after exiting PSM mode?
My modem configuration:
```
AT+CFGRI=1
AT+CURCCFG="QUALCOMM",1
AT+CURCCFG="SMS",1
AT+CURCCFG="SYS",1
AT+CURCCFG="TCPIP",0
AT+CURCCFG="NETWORK",0
AT+CPSMSTATUS=1
AT+CPSMS=1,,,"10100010","00000011"
```

lewisxhe, 2023-05-27:
PSM mode The wake-up time needs to be confirmed with the operator

closed 2023-05-27 by lewisxhe

4ib3r, 2023-05-29:
PSM mode and timers are working, modem wake-up in time and status led turn on after this event, when esp is not in depth sleep mode, RI pin change state regularly, but when I turn on depth sleep in esp and set `esp_sleep_enable_ext0_wakeup(GPIO_NUM_3, 0);` to this pin it not working, and pin don't change state.

lewisxhe, 2023-05-30:
Use `esp_sleep_enable_ext1_wakeup(GPIO_NUM_3,LOW)`

4ib3r, 2023-06-09:
I found the problem. The transitor based level converter for the RI pin gets 1.8V from the vdd_ext modem pin and this voltage is missing when the modem goes into PSM mode. When the modem wakes up the RI and vdd_ext drop to 1.8V at the same time but nothing changes in this case for esp32 on GPIO 3.