Xinyuan-LilyGO/LilyGo-T-SIM7080G #25: GPS example not working properly

gfilippi, 2023-08-10:
I'm using the T-SIM7080G board with nothig attached  to the external gpio, only the board itself with the usbc cable.

I'm testing the tiny gsm example "as is" (fixed the define for the PMU which was changed recently)

I'm facing the same issue described here:
https://github.com/Xinyuan-LilyGO/LilyGO-T-SIM7000G/issues/98

===
In my case (7080, example from this repo as-is)

1) I'm in open air
2) I run the example once and I got almost immediate GPS link (!)
3) reset the board .. does not get a link anymore.
:-(

I've re-flashed the board with the same example code a couple of times ... but nothing.

The code seems to loop forever searching on the "AT+CGNSINF".

===
any idea how to debug this?
is there anything written permanently in the simcom device which stops the GPS for finding the link the second time we run the example?

===
was the example tested with the current mainline?

lewisxhe, 2023-11-04:
Really? We did not add the COLD/WARM/HOT startup command in the test, but it was correctly located. Have you sent other commands to change the settings?

lewisxhe, 2024-01-19:
If you have other questions, please reopen

closed 2024-01-19 by lewisxhe