Xinyuan-LilyGO/LilyGo-T-SIM7080G #26: please clarify 24/7 operation with solar and PMU

gfilippi, 2023-08-11:
It is not clear to me how the device can sustain 24.7 (i.e. continuous) working conditions using the batter and a solar panel.

If I understand correctly the behavior of the PMU if we enter the condition where there is no sun for a long time and the battery goes to the minimum level the device will shut down.

at this point how do we return to functionality ?

- the PMU will not turn ON if we do not press the BOOT button
- the battery is not charging if the PMU is not turned ON
- if the battery is low ... none of the above will work.

===
Suppose this is a remote sensor .. how do we guarantee that the device will turn ON when the sun comes back?

what am I missing?
:-)

gfilippi, 2023-08-12:
I did study the datasheet of the PMU and looked at the schematics of the board.

It seems to me that the solar panel (VSOLAR) is used ONLY to charge the battery (page 5 of the schematics) and that we have a fet to override charging when there is VBUS connected.

As I understand it .. this confirms the issue: if the battery goes on depletion at night ... the solar panel will charge the battery the day after .. but the board will NEVER turn on by itself even if the battery is fully charged because we need to press the PWRON button on the PMU.

how to solve this problem to allow for a 24/7 operational behavior of the device??

also .. why do we need the CN3065 ? it seems to me that the AXP2101 pmu is capable of charging the battery, isn't it?
If this is true... why not using the VSOLAR as an alternative to the VBUS itself (with a proper 5V regulator on VSOLAR of course..)

please help

lewisxhe, 2023-08-12:
The PMU cannot be automatically turned on at present, you can only handle it through software.
For example, you should check whether the voltage is lower than the normal working voltage value at intervals, if it is lower, then set the esp32 to sleep state, and use the timing wake-up function to check the voltage state at intervals to see if it can work normally.

VSOLAR will charge the battery regardless of whether the PMU is powered on or not.

PMU cannot directly use solar energy for charging with high efficiency. Use CN3065 is this chip has automatic boost converter, it is the best choice for solar charging

gfilippi, 2023-08-17:
ok, understood.

but still .. even if we sleep the battery will discharge .. and in a very long (with no sun) we will enter a situation where the board will not boot anymore even if the battery is charged when the sun comes back.

what am I missing?

gfilippi, 2023-08-20:
Hi,
we have clarified that CN3065 will always charge the battery, and from the schematic this is happening from both Vsolar and VBus.

so what is the need of setting PMU charging limits (see the minmal pmu example) if the pmu is not involved in controlling the battery?

is this mandatory for a correct behavior or just a test of the API?

lewisxhe, 2023-08-20:
1. When the battery voltage drops below 3.4V, the PMU will switch to the shutdown state, and it cannot restart automatically at this time. This is the limitation of the PMU. If you need to restart the power, you must manually press the PWR button.

2. Set charging limit only set USB charging, solar charging is controlled by CN3065, can not be programmed, this is not mandatory, just for demonstration

lewisxhe, 2024-01-19:
If you have other questions, please reopen

closed 2024-01-19 by lewisxhe