Xinyuan-LilyGO/LilyGo-T-SIM7080G #28: I can't connect to the GPRS network

MrRocklion, 2023-08-26:
I'm trying to test the MinimalModemNBIOT example but for some reason after I correctly configured the APN, the only message I get is the one below.

OK
AT+CGREG?

+CGREG: 0,2
I am powering the board directly from the usb of my laptop. I tried the GPS test and it worked correctly but when I try to test the GPRS test it can't connect, I already tried with different operators and sim cards and the result is the same.
[image]

lewisxhe, 2023-11-04:
It seems that the SIM card is inserted backwards?
[image]

aovestdipaperino, 2024-03-07:
I have the same issue and the SIM is installed correctly (SMS READY)
But it goes from 
```
OK
AT+CGREG?

+CGREG: 0,2
```
to
```
OK
AT+CGREG?

+CGREG: 0,3
```

Signal quality is top (31), GPS works very well, tried different SIMs, `AT+COPS?` returns 0.
I uploaded the latest firmware provided here (B08) with no changes.

lewisxhe, 2024-03-07:
@aovestdipaperino 
APN Settings  https://github.com/Xinyuan-LilyGO/LilyGo-T-SIM7080G/commit/9ea916e3412e8faa9db61fa6a49753919dec61dc

aovestdipaperino, 2024-03-07:
Nope, they are correct.
People at SIMCOM explained me that this module doesn't support LTE/4G, but only NB-IOT or CATM.
And this support is a function of the SIM.
So basically you can't plug in a phone SIM and expect to work, but special SIMs are required.
Finding those, since information is scarse, is hard.
I am getting one delivered today and I will report back.

lewisxhe, 2024-03-07:
Yes, SIM7080 does not support LTE. You must use a SIM card that supports NB or CAT-M to work.

closed 2024-05-01 by github-actions[bot]