Xinyuan-LilyGO/LilyGo-T-SIM7080G #30: No external power supply possible

Klabbob, 2023-09-24:
Am I correct in the assumption that it is not possible to power the board with an external power supply other than the USB port? There is no pin assigned to VBUS on the header and DC1, DC5 and VSYS pins are only outputs, rights?

lewisxhe, 2023-11-04:
If you are sure you don't need to use the USB input, you can use the 5V input from VSYS, but you cannot connect a battery.

Klabbob, 2023-11-05:
If I connect 5V to VSYS, the board will not turn on and you als write in your tips, that it is only an output, not an input. How can I power the board through the pin header?

> 3. The board VSYS is the internal 5V boost output of the PMU, which can only be output but not input. Please do not connect an external power supply load greater than 500mA

lewisxhe, 2024-01-19:
Has this issue been resolved?

Klabbob, 2024-01-19:
No, please read the last message from me. 

VSYS is only an output, not an input --> you can not power the board by connecting a external power supply to VSYS. I have also tested this to verify it.

Can you confirm that it is **not** possible to power the board with an external power source trough the header? Only USB C and the Battery Connector can be used, correct?

lewisxhe, 2024-01-19:
You can access the 5V input from here, which is the 5V access point for USBC.

VSYS is the battery path output terminal. You can use it as an input or as an output. The voltage of VSYS depends on your input voltage. You can also directly connect 5V from VSYS, but this means there will be no Automatic switching function, and cannot be connected to power from battery or USBC
[image]

Peer77, 2024-03-28:
I can confirm, that connecting 5V to the pad with the red arrow powers up the board and charging is working. It would be great to have a dedicated pin on the pinheader, as it is not ideal to solder direct at the diode.

closed 2024-05-12 by github-actions[bot]