Xinyuan-LilyGO/LilyGo-T-SIM7080G #32: PMU failure after some time

gfilippi, 2023-10-10:
I have n.5 units that I was testing in 24/7 mode and one unit failed today. 

it was charging the battery (which is now full) but when I checked the board it was "warm" and it is not powering up anymore.
even if I press the power-on button I do not see the 3.3v coming up ... and I do have now 5v from USB (I removed the battery)

how to debug this? how to check if the pmu is dead?

thanks

lewisxhe, 2023-11-04:
Run this [MinimalPowersExample](https://github.com/Xinyuan-LilyGO/LilyGo-T-SIM7080G/blob/master/examples/MinimalPowersExample/MinimalPowersExample.ino)to see if it works properly and outputs

lewisxhe, 2024-01-19:
If you have any other questions, please reopen

closed 2024-01-19 by lewisxhe