Xinyuan-LilyGO/LilyGo-T-SIM7080G #33: Device not in Verizon's DMD. Unable to activate.

danmandle, 2023-10-24:
The product specs says that this device is certified to be on the Verizon cellular network but the device IMEI is not in Verizon's DMD (Device Management Database). Device manufacturers need to upload the IMEIs of compatible devices so that they can be activated on the cellular network. 

Here are the details for how to do that. See Section 4: https://opendevelopment.verizonwireless.com/content/dam/opendevelopment/pdf/OpenAccessReq/OD_Certification_Process_v33.pdf

lewisxhe, 2023-11-04:
You can send `AT+CGSN` to check whether the returned IMEI number is consistent with the IMEI column of the module label

danmandle, 2023-11-04:
That's not my issue. The IMEIs of the modules have not been submitted to Verizon. This prevents the devices from being activated on Verizon's network.

lewisxhe, 2023-11-07:
What I mean is, can you first check whether the IMEI returned by the module software is consistent with the module label? It is possible that the IMEI in the software is incorrect.

Regarding whether I can use Verizon, I still need to check with SIMCOM.

danmandle, 2023-11-07:
> Regarding whether I can use Verizon, I still need to check with SIMCOM.
It is listed on the spec sheet that this device is certified to be on the Verizon network

lewisxhe, 2023-11-09:
I consulted SIMCOM. The IMEI number of each device has been registered with GSM and can be queried at https://www.imei.info/. It has been certified by Verizon and has been registered.

lewisxhe, 2024-01-19:
If you have any other questions, please reopen

closed 2024-01-19 by lewisxhe

tadenham, 2024-09-27:
I am having the exact same issue with a SIM7070G. Unable to activate because the IMEI is not in Verizon's DMD. Not being able to use Verizon SIMs will be very limiting...