Xinyuan-LilyGO/LilyGo-T-SIM7080G #34: simcom 7080 link quality variance from board to board

gfilippi, 2023-10-29:
I have a batch of 6 board where 4 are working fine and 2 of them are having issues to connect to the network.

when they connect ... I see a poor signal (AT+CSQ is reporting 19,99 which means belo -100dBm ... very low signal)
the other boards (in the same spot of my room, using the exact same antenna from the lilygo kit) work fine with a signal of about -70dBm

I tried to swap the antenna with one of a working board but the result is the same, very poor signal.

==
I do not see anything special on the schematic (just to 0-ohm resistor to the uFL connector) and the PCB seems done very well to me, with a proper shielding on the RF antenna connection.

1) is this an issue on the SIMCOM firmware? 
 2) is there anyone seeing the same variability of RF signal from board to board ?

gfilippi, 2023-10-29:
I take it back: CSQ=99 means about -70dBm .. like for the other boards.

so I have no clue why these two modules take forever to get registered on the network in comparison to the other four. I need to wait for MINUTES while for the other boards in 20-30sec I do have link.
:-(

lewisxhe, 2023-11-04:
Try sending `AT+SIMCOMATI` to see if the returned software version is consistent, and see if it is due to differences in software versions.

gfilippi, 2023-11-17:
hello, after debugging the issue was related to the list of channles to scan (which was including ALL the possible channels). I reduced the list to the one supported locally and now it seems a lot more stable

gfilippi, 2023-11-17:
not a bug

closed 2023-11-17 by gfilippi