Xinyuan-LilyGO/LilyGo-T-SIM7080G #38: LilyGo-T-SIM7080G won't turn-on with battery only

xfiber, 2024-01-13:
Dear community, 

I am facing an issue with LilyGo-T-SIM7080G when connected to battery as only power source.
When battery is connected I am not able to turn-on the board.
I've tried ready this:

_"If the power supply of the board cannot be turned on when the battery is connected, please check that the mechanical switch beside the USB is turned to ON, and then press the board PWR for 2 seconds. At this time, the board should start to work"._ 

This does not help at all. Board does not want to turn on.

However, whenever I turn-on the board using USB-C power, and later unplug power from USB-C, board will stay on using battery power source. Whenever I turn off the board and try to start with only battery source connected, board will not turn-on. I always need to "jump start" using usb-c power.

I would appreciate your support.
Thank you.

lewisxhe, 2024-01-15:
It looks like USB CDC is blocked from starting. Try setting USB CDC ON BOOT = DISABLE
[image]

mathieucarbou, 2024-01-17:
@lewisxhe : what is this interface? I am starting a project with these boards and I will have the same need. Thanks:-)

mathieucarbou, 2024-01-17:
> @lewisxhe : what is this interface? I am starting a project with these boards and I will have the same need. Thanks:-)

Looking at the read me, this is Arduino IDE? Do we have some equivalent flags to pass to the compiler when we use platformio?

Thanks!

lewisxhe, 2024-01-18:
@mathieucarbou  I already have a note in the platformio.ini file
https://github.com/Xinyuan-LilyGO/LilyGo-T-SIM7080G/blob/07233c2de91e3ee80e98b61416066ef2e677465b/platformio.ini#L56C13-L56C13

xfiber, 2024-01-18:
> It looks like USB CDC is blocked from starting. Try setting USB CDC ON BOOT = DISABLE [...]

Thank you! this solved the issue.
However, when this is set as above I am not able to monitor serial port. Is this correct behaviour ?

xfiber, 2024-01-18:
> > @lewisxhe : what is this interface? I am starting a project with these boards and I will have the same need. Thanks:-)
> 
> Looking at the read me, this is Arduino IDE? Do we have some equivalent flags to pass to the compiler when we use platformio?
> 
> Thanks!

Correct. Framework is Arduino IDE.

lewisxhe, 2024-01-19:
CDC controls whether to use USBC as print output. If set to Disable, the default Serial output will be directed to GPIO43, GPIO44. This is the QWIIC interface near the SIM7080 battery holder.

closed 2024-04-30 by github-actions[bot]