Xinyuan-LilyGO/LilyGo-T-SIM7080G #41: Manually selecting an operator fails

mathieucarbou, 2024-03-07:
Both with CAT-M and NB-IOT, if I select an operator, for example like that:

```
AT+COPS=1,0,"F SFR",7
```

It always fails with `ERROR`.

Only `AT+COPS=0` works.

The problem is that I would like to select the operator because I need to exclude one (which is blocking my IoT provider).

Please let me know how I can select the operator.

mathieucarbou, 2024-03-08:
@lewisxhe  ?

lewisxhe, 2024-03-08:
Looks like a format error, try
```
AT+COPS=1,2,"20810"
```

mathieucarbou, 2024-03-08:
> Looks like a format error, try
> 
> ```
> AT+COPS=1,2,"20810"
> ```

No it does not help... Here is the full command list:

```
> AT
OK

# Preferred mode

> AT+CNMP=?
+CNMP: ((2-Automatic),(13-GSM Only),(38-LTE Only),(51-GSM And LTE Only))
> AT+CNMP=2

# Preferred IoT mode

> AT+CMNB=?
+CMNB: ((1-Cat-M),(2-NB-IoT),(3-Cat-M And NB-IoT))
> AT+CMNB=3

$ Set bands for France

> AT+CBANDCFG=?
+CBANDCFG: (CAT-M,NB-IOT),(1,2,3,4,5,8,12,13,14,18,19,20,25,26,27,28,66,71,85)

# https://en.wikipedia.org/wiki/List_of_LTE_networks_in_Europe
> AT+CBANDCFG="CAT-M",1,3,8,20,28

# https://www.everythingrf.com/community/nb-iot-frequency-bands-in-europe
> AT+CBANDCFG="NB-IOT",3,8,20

# Operator selection: try to manually set SFR

> AT+COPS=?
+COPS: (1,"F SFR","SFR","20810",9),(1,"F SFR","SFR","20810",7),(3,"F-Bouygues Telecom","BYTEL","20820",7),,(0,1,2,3,4),(0,1,2)

> AT+COPS=1,2,"20810"
ERROR
```

mathieucarbou, 2024-03-08:
Sme for `AT+COPS=1,2,"20820"`.

If an operator, let's say SFR or Bouygues here, is blocking my M2M / IoT SIM provider, could this create the error ?

mathieucarbou, 2024-03-08:
Ok! it worked! With: `AT+COPS=1,2,"20810",7`

mathieucarbou, 2024-03-08:
With Onomondo in France:

```
> AT
OK

# Preferred mode

> AT+CNMP=?
+CNMP: ((2-Automatic),(13-GSM Only),(38-LTE Only),(51-GSM And LTE Only))
> AT+CNMP=2

# Preferred IoT mode

> AT+CMNB=?
+CMNB: ((1-Cat-M),(2-NB-IoT),(3-Cat-M And NB-IoT))
> AT+CMNB=3

$ Set bands for France

> AT+CBANDCFG=?
+CBANDCFG: (CAT-M,NB-IOT),(1,2,3,4,5,8,12,13,14,18,19,20,25,26,27,28,66,71,85)

# https://en.wikipedia.org/wiki/List_of_LTE_networks_in_Europe
> AT+CBANDCFG="CAT-M",1,3,8,20,28

# https://www.everythingrf.com/community/nb-iot-frequency-bands-in-europe
> AT+CBANDCFG="NB-IOT",3,8,20

# Operator selection: try to manually set SFR

> AT+COPS=?
+COPS: (1,"F SFR","SFR","20810",9),(1,"F SFR","SFR","20810",7),(3,"F-Bouygues Telecom","BYTEL","20820",7),,(0,1,2,3,4),(0,1,2)

> AT+COPS=1,2,"20810",7

# APN

> AT+CGDCONT=1,"IP","onomondo"

> AT+CNCFG=0,1,"onomondo"

# Check status

> AT+CEREG?
+CEREG: 0,5

# Check used Operator

> AT+COPS?
+COPS: 1,2,"20810",7

# Activate DATA

> AT+CNACT=0,1
> AT+CNACT?
+CNACT: 0,1,"100.83.225.165"

# Ping

> AT+SNPDPID=0

> AT+SNPING4="8.8.8.8",5,16,5000
+SNPING4: 1,8.8.8.8,60000
+SNPING4: 2,8.8.8.8,2787
+SNPING4: 3,8.8.8.8,216
+SNPING4: 4,8.8.8.8,174
+SNPING4: 5,8.8.8.8,245
```

mathieucarbou, 2024-03-08:
Thanks :-)
I am closing.

closed 2024-03-08 by mathieucarbou