Xinyuan-LilyGO/LilyGo-T-SIM7080G #42: EnableGPS failed after sending AT command

tuando90, 2024-03-10:
Hi Team,

I am facing an issue with `MinimalModemGPSExample.ino` example.

When trying to change GNSS accuracy to Only High Accuracy If I change to code inside `#if 0 block` into:
```c++
#if 1
    /*
    ! GNSS Work Mode Set
     <gps mode> GPS work mode.
        1 Start GPS NMEA out.
    <glo mode> GLONASS work mode.
        0 Stop GLONASS NMEA out.
        1 Start GLONASS NMEA out.
    <bd mode> BEIDOU work mode.
        0 Stop BEIDOU NMEA out.
        1 Start BEIDOU NMEA out.
    <gal mode> GALILEAN work mode.
        0 Stop GALILEAN NMEA out.
        1 Start GALILEAN NMEA out.
    <qzss mode> QZSS work mode.
        0 Stop QZSS NMEA out.
        1 Start QZSS NMEA out.*/

    //GNSS Work Mode Set GPS+BEIDOU
    modem.sendAT("+CGNSMOD=1,1,0,0,0");
    modem.waitResponse();

    /*
    GNSS Command,For more parameters, see <SIM7070_SIM7080_SIM7090 Series_AT Command Manual> 212 page.
    <minInterval> range: 1000-60000 ms
     minInterval is the minimum time interval in milliseconds that must elapse between position reports. default value is 1000.
    <minDistance> range: 0-1000
     Minimum distance in meters that must be traversed between position reports. Setting this interval to 0 will be a pure time-based tracking/batching.
    <accuracy>:
        0  Accuracy is not specified, use default.
        1  Low Accuracy for location is acceptable.
        2 Medium Accuracy for location is acceptable.
        3 Only High Accuracy for location is acceptable.
    */
    // minInterval = 1000,minDistance = 0,accuracy = 0
    modem.sendAT("+SGNSCMD=2,1000,0,3");
    modem.waitResponse();

    /*
    // Turn off GNSS.
    modem.sendAT("+SGNSCMD=0");
    modem.waitResponse();
    */
    
#endif
    delay(500);

    // GPS function needs to be enabled for the first use
   while(1) {
      if (modem.enableGPS() == false) {
          Serial.print("Modem enable gps function failed!!");
          delay(5000);
      }
    }
```

Then `modem.enableGPS() ` often failed. I added the code to retry to enableGPS but sometimes it works and even if it works it takes long time to wait.

Is there any suggestion to fix this problem?

Thanks in advance!

lewisxhe, 2024-03-10:
modem.enableGPS() returns false?

Did you know that the positioning function cannot be used when the SIM7080 network is turned on?

tuando90, 2024-03-13:
> modem.enableGPS() returns false?
> 
> Did you know that the positioning function cannot be used when the SIM7080 network is turned on?

@lewisxhe  Yes, it returns false. I retry the enableGPS() function several times then after some time it works.
I just used GPS and BLE.
Do you have any idea?

lewisxhe, 2024-03-13:
oh ! Are you saying that after changing #if 0 to #if 1, calling modem.enableGPS() will return failure? oh ! I forgot about this. After running the SGNSCMD command, CGNSPWR will be rejected. The two commands cannot be used together. This was my mistake. I forgot to delete it. You can test it according to this application manual.

https://github.com/Xinyuan-LilyGO/LilyGo-T-SIM7080G/blob/master/datasheet/SIM7070_SIM7080_SIM7090%20Series_GNSS_Application%20Note_V1.02.pdf

tuando90, 2024-03-13:
Oh I see. 
I tried what you suggested and it works well now.
Thanks @lewisxhe.

tuando90, 2024-03-13:
Hi @lewisxhe 
Do you mean I should change the setup code like this?

```c++
void setup()
{

    Serial.begin(115200);

    //Start while waiting for Serial monitoring
    while (!Serial);

    delay(3000);

    Serial.println();

    // Setup BLE
    setupBLE();

    /*********************************
     *  step 1 : Initialize power chip,
     *  turn on modem and gps antenna power channel
    ***********************************/
    if (!PMU.begin(Wire, AXP2101_SLAVE_ADDRESS, I2C_SDA, I2C_SCL)) {
        Serial.println("Failed to initialize power.....");
        while (1) {
            delay(5000);
        }
    }
    //Set the working voltage of the modem, please do not modify the parameters
    PMU.setDC3Voltage(3000);    //SIM7080 Modem main power channel 2700~ 3400V
    PMU.enableDC3();

    //Modem GPS Power channel
    PMU.setBLDO2Voltage(3300);
    PMU.enableBLDO2();      //The antenna power must be turned on to use the GPS function

    // TS Pin detection must be disable, otherwise it cannot be charged
    PMU.disableTSPinMeasure();

    /*********************************
     * step 2 : start modem
    ***********************************/

    Serial1.begin(115200, SERIAL_8N1, BOARD_MODEM_RXD_PIN, BOARD_MODEM_TXD_PIN);

    pinMode(BOARD_MODEM_PWR_PIN, OUTPUT);

    digitalWrite(BOARD_MODEM_PWR_PIN, LOW);
    delay(100);
    digitalWrite(BOARD_MODEM_PWR_PIN, HIGH);
    delay(1000);
    digitalWrite(BOARD_MODEM_PWR_PIN, LOW);

    int retry = 0;
    while (!modem.testAT(1000)) {
        Serial.print(".");
        if (retry++ > 15) {
            // Pull down PWRKEY for more than 1 second according to manual requirements
            digitalWrite(BOARD_MODEM_PWR_PIN, LOW);
            delay(100);
            digitalWrite(BOARD_MODEM_PWR_PIN, HIGH);
            delay(1000);
            digitalWrite(BOARD_MODEM_PWR_PIN, LOW);
            retry = 0;
            Serial.println("Retry start modem .");
        }
    }
    Serial.println();
    Serial.print("Modem started!");

    /*********************************
     * step 3 : start modem gps function
    ***********************************/

    //  When configuring GNSS, you need to stop GPS first
    // modem.disableGPS();
    delay(500);

#if 1
    /*
    ! GNSS Work Mode Set
     <gps mode> GPS work mode.
        1 Start GPS NMEA out.
    <glo mode> GLONASS work mode.
        0 Stop GLONASS NMEA out.
        1 Start GLONASS NMEA out.
    <bd mode> BEIDOU work mode.
        0 Stop BEIDOU NMEA out.
        1 Start BEIDOU NMEA out.
    <gal mode> GALILEAN work mode.
        0 Stop GALILEAN NMEA out.
        1 Start GALILEAN NMEA out.
    <qzss mode> QZSS work mode.
        0 Stop QZSS NMEA out.
        1 Start QZSS NMEA out.*/

    //GNSS Work Mode Set GPS+BEIDOU
    modem.sendAT("+CGNSMOD=1,1,0,0,0");
    modem.waitResponse();

    /*
    GNSS Command,For more parameters, see <SIM7070_SIM7080_SIM7090 Series_AT Command Manual> 212 page.
    <minInterval> range: 1000-60000 ms
     minInterval is the minimum time interval in milliseconds that must elapse between position reports. default value is 1000.
    <minDistance> range: 0-1000
     Minimum distance in meters that must be traversed between position reports. Setting this interval to 0 will be a pure time-based tracking/batching.
    <accuracy>:
        0  Accuracy is not specified, use default.
        1  Low Accuracy for location is acceptable.
        2 Medium Accuracy for location is acceptable.
        3 Only High Accuracy for location is acceptable.
    */
    // minInterval = 1000,minDistance = 0,accuracy = 0
    modem.sendAT("+SGNSCMD=2,1000,0,0");
    modem.waitResponse();

    // Turn off GNSS.
    /*
    modem.sendAT("+SGNSCMD=0");
    modem.waitResponse();
    */
#endif
    delay(500);

    // GPS function needs to be enabled for the first use
    /*
    if (modem.enableGPS() == false) {
        Serial.print("Modem enable gps function failed!!");
        while (1) {
            delay(5000);
        }
    }
    */
}
```

lewisxhe, 2024-03-13:
Use ATDebug first, follow the instructions in the PDF linked above, and try it

closed 2024-04-30 by github-actions[bot]