Xinyuan-LilyGO/LilyGo-T-SIM7080G #47: The battery has stopped working

manzajaro, 2024-03-27:
Hi, after several days of testing a code and battery life, it turns out that this morning the battery-only power supply stopped working and I don't understand why.

I'm compiling with these options!

[image]

I've taken measurements with the multimeter and I can see that the PMU is getting 4.1V.

When I run the code connected to the usb cable it works correctly, but as soon as I remove it the device switches off.

The power button works correctly, because when it is powered by usb I press it for X seconds and the device turns off.

I run the following code [MinimalPowersExample](https://github.com/Xinyuan-LilyGO/LilyGo-T-SIM7080G/blob/master/examples/MinimalPowersExample/MinimalPowersExample.ino) to see if the PMU is OK and it returns the following result.

[image]

Can someone help me, thanks

lewisxhe, 2024-03-29:
[image]
Is the switch turned on?

manzajaro, 2024-03-29:
Hi @lewisxhe,

The button is on, the battery is on and I have been able to check that the PMU receives 4.1v from the battery.

https://github.com/Xinyuan-LilyGO/LilyGo-T-SIM7080G/assets/6926926/30cd97de-9abf-46dc-b8fc-19a7ff6d2fd1

lewisxhe, 2024-03-29:
Can you follow the instructions here and remove the diode and then run the MinimalPowersExample to see if the battery prints a connection? Now due to the presence of this diode, the battery connection will print even if the battery is not inserted.

https://github.com/Xinyuan-LilyGO/LilyGo-T-SIM7080G/issues/49

manzajaro, 2024-03-29:
> Can you follow the instructions here and remove the diode and then run the MinimalPowersExample to see if the battery prints a connection? Now due to the presence of this diode, the battery connection will print even if the battery is not inserted.

I'll do it now and tell you the result.

This is the consumption of the board, when it is switched on and the pwr button is pressed.

[image]

Thanks @lewisxhe

manzajaro, 2024-03-29:
Logs with the diode removed, as you can see it detects well. 1st logs without battery, then inserted and detected.

[image]

lewisxhe, 2024-03-30:
Can detect battery insertion and removal, but cannot use battery power?

manzajaro, 2024-03-30:
Correct, the battery alone doesn't start. You've already seen the power consumption measurements of the board.

It stopped working on its own, it was undergoing a test throughout the whole night and the next morning the board wasn't functioning and the battery was charged.

lewisxhe, 2024-04-02:
As you said, the battery path inside the AXP2101 chip may have been disconnected. Please contact your dealer for after-sales service.

closed 2024-04-05 by manzajaro