Xinyuan-LilyGO/LilyGo-T-SIM7080G #48: No ">" prompt after +CASEND after wakeup from sleep (DRX)

btuysser, 2024-03-27:
My project is based on [MinimalModemNBIOTExample](https://github.com/Xinyuan-LilyGO/LilyGo-T-SIM7080G/tree/master/examples/MinimalModemNBIOTExample). I am transmitting a ~28 byte UDP datagram every 60s and I want the modem to sleep. I am using DRX as opposed to eDRX or PSM due to future requirements.I added some UDP code (as that is not supported by TinyGSM), set +SCLKC=1 and I'm controlling DTR. 

In the setup function I acquire NB-IOT, register and open a UDP connection. 

Here is my loop function and a modifed TinyGsmSim7080::modemSend :  

```
 int16_t modemSend(const void* buff, size_t len, uint8_t mux) {

    while(waitResponse(5,GF(">")) != 1) {
      sendAT(GF("+CASEND="), mux, ',', (uint16_t)len);
    }

    stream.write(reinterpret_cast<const uint8_t*>(buff), len);
    stream.flush();

    // OK after posting data
    if (waitResponse() != 1) { return 0; }

    return len;
  }

void udpSend(const String& msg)
{
    // keep sending if not successful
    modem.modemSend(msg.c_str(), msg.length(), 0);

    modem.sendAT("+CRAI=1");
    modem.waitResponse(5);
}

void loop()
{
    static int cnt = 0;
    long t0 = millis();
    // pull DTR low in case we were sleeping
    digitalWrite(BOARD_MODEM_DTR_PIN, LOW);
    // use +CASEND to send UDP datagram and set +CRAI=1
    udpSend("012345678 UDP test Count " + String(cnt++, DEC));
    // we can go back to sleep now
    digitalWrite(BOARD_MODEM_DTR_PIN, HIGH);
    Serial.printf("\n\r--------------  SLEEP AFTER %d --------------\n\r", millis() - t0);
    delay(60000);
}
```
When I call +CASEND immediately after wakeup there is no ">" prompt on the modem serial connection. I added retries in the modemSend method and with a timeout of 5ms it takes 3 tries to get a ">" prompt. This results in an awake time of ~24ms instead of an ideal ~15ms. When increasing the waitResponse timeout to 50ms I would expect the number of tries to go down but it remains between 1 and 3 tries for an awake time of 65-117ms. 

Another workaround is to add a 300ms delay after pulling DTR low but the impact on awake time is unacceptable. 

Log : 
```
AT+CPSI?
+CPSI: LTE NB-IOT,Online,310-260,0xF300,7573966,450,EUTRAN-BAND4,2038,0,0,-11,-98,-87,13
OK

AT+CSCLK=1
OK
AT+CACLOSE=0

AT+CACID=0
OK

AT+CAOPEN=0,0,"UDP","<IP>",<PORT>
OK

AT+CASEND=0,26
>012345678 UDP test Count 0 
OK
AT+CRAI=1
OK

--------------  SLEEP AFTER 21 ms --------------
AT+CASEND=0,26
AT+CASEND=0,26
AT+CASEND=0,26
>012345678 UDP test Count 1 
OK
AT+CRAI=1
OK
--------------  SLEEP AFTER 32 ms --------------
AT+CASEND=0,26
AT+CASEND=0,26
AT+CASEND=0,26
>012345678 UDP test Count 2 
OK
AT+CRAI=1
OK
--------------  SLEEP AFTER 32 ms --------------
.....

```

btuysser, 2024-03-27:
Firmware Revision : 1951B12SIM7080

lewisxhe, 2024-03-29:
Are you referring to the wake-up reaction time being too long?

btuysser, 2024-03-29:
Since there is almost no information in the low power application note I hoped that the wakeup reaction time would be  be transparent to the user. After pulling DTR low I expected the modem to execute any serial command after finishing its own internal wakeup sequence. 

If that is not the case then how does a user determine the **earliest** time it's safe to interact with the modem (in this case sending a UDP packet via +CASEND)?

btuysser, 2024-04-16:
Would it be possible to extend the MinimalModemSleepMode.ino example by doing multiple sleeps? 

My current sequence is 

- pull DTR low 
- send AT commands with a timeout of 1ms until OK is received or a threshold is exceeded
- +CASEND
- pull DTR high
- .....

I just don't know if this is the optimal sequence.

closed 2024-06-01 by github-actions[bot]