Xinyuan-LilyGO/LilyGo-T-SIM7080G #49: Set charging current limit has no effect

Peer77, 2024-03-28:
Hi Lewis!

Thank you for your good work in developing the AXP libraries. I would love to get the LilyGo SIM7080G module up and running exactly as it was possible when using the SIM800L with the AXP192  

The ESP32 with SIM800 and AXP192 is showing perfect operation:

# 100mA is set and in total ca. 180mA seen on LabPSU (must be 100mA charging current and rest accounts for the ESP to keep everything below 250mA.

But when using the same battery with the SIM7080G and also set 100mA for charging current, the drawn power is always seen as 500mA (except in the end when it is nearly charged, the drawn current drops.)

I also tried to set precharge current and termination current to 0mA -> without success.

Issue could not get resolved by switching  off unused powerlines as per below:

 // Modem 2700~3400mV VDD
    PMU.setDC3Voltage(3000);
    PMU.enableDC3();

    // Modem GPS Power
    PMU.setBLDO2Voltage(3300);
    PMU.enableBLDO2();

    // DC1 is untouched, as it is needed!

    // switch off unused powerlines
    PMU.disableDC2();
    PMU.disableDC4();
    PMU.disableDC5();

    PMU.disableALDO1(); // CAM 
    PMU.disableALDO2(); // CAM
    PMU.disableALDO3(); // SDCARD
    PMU.disableALDO4(); // CAM
    PMU.disableBLDO1(); // PURPOSE ???

    PMU.disableCPUSLDO(); // PURPOSE ???
    PMU.disableDLDO1(); // PURPOSE ???
    PMU.disableDLDO2(); // PURPOSE ???

The current is set to be limited to 100mA via 
PMU.setChargerConstantCurr(XPOWERS_AXP2101_CHG_CUR_100MA)

checked: 
uint8_t val = PMU.getChargerConstantCurr();

But still the drawn current is seen as ca. 500 mA when the battery is NOT charged.
When a full charged battery is connected, the current of the entire ESP32S3 is seen at 70mA total.

Lewis, could you please confirm if the charging current in xpowers_axp2101_chg_curr_t are working for you?

Greetings! Peer

lewisxhe, 2024-03-29:
oh ! I verified it this morning, and I found the problem that the current cannot be set. This is a hardware problem. You can remove the diode in the picture, so that the charging current will not charge from CN3065, but from AXP2101.

[image]

At the same time, I found a software BUG. Failure to change the register will cause the charging current setting to be incorrect.

thank you for your feedback

Peer77, 2024-04-03:
Hi Lewis,

yes, after thorough testing I can confirm your advice fixed the issues, thanks alot!

I did:

a) remove the big diode next to the USB-connector and
b) installed the XPowersLib (ver. 0.2.2)

Result:

First I have set 300mA as charging current then checked the readings.

At last I have set 100mA as charging current -> the measured current is pretty much the same as it was with the beloved AXP192 (100mA chg current + 80mA for the board)

Now after the diode has moved away from the Lilygo PCB, it is much easier to connect a small cable to the pad with the red arrow to apply 5V as charging voltage. As soon as 5V gets connected the board switches on and charges the battery. When the 5V is disconnected the board remains switched on (when battery is inserted)

To power off the board completely I use the 5V for a MOSFET to interrupt the negative pole of the 18650 battery.

If there is a future version, it would be very helpful when there is a dedicated Vin pin on the headers (replacing the VSYS pin, e.g??) If I remember that correctly, it is only possible to pickup 5V from that VSYS pin or feed 5V into the boad, but battery is not being charged by VSYS voltage.

As it is fully resolved, the issue can be closed.

Thanks again!

lewisxhe, 2024-04-29:
I will revise the later versions based on the feedback from the first version, thank you.

closed 2024-04-29 by lewisxhe

Peer77, 2024-04-29:
Hi Lewis,

thank you. Commit 64304d8 in combination with the removed diode works very well. Thanks for your help!

smehner1, 2024-11-20:
Hi @Peer77 

can you please explain a little more how to do that? Do you have a schematic?

> To power off the board completely I use the 5V for a MOSFET to interrupt the negative pole of the 18650 battery.

Best,

Stefan