Xinyuan-LilyGO/LilyGo-T-SIM7080G #55: Correct drivers for Windows 11

sporkyduck, 2024-04-11:
Hi,

Newbie here. Posting here as community seems to be down for maintenance. 

What is the correct selection of drivers needed to setup USB connection to the board on Windows 11? I find no references anywhere here or on the website. However, found these, that seems to be working somewhat when I follow the instructions: https://www.waveshare.com/wiki/SIM7080G_Cat-M/NB-IoT_HAT. Three ports are created

However, there is no indication on which one of the three ports created that should be used for actual connection in Arduino IDE?
Are there any instructions on drivers and port installations on Windows 11? Any information or tutorials I find starts when this is already set up.

Thanks for any tips/pointers! Cheers

lewisxhe, 2024-04-11:
Upload sketches using the motherboard's USB-C port, not Micro-USB

sporkyduck, 2024-04-11:
Ok, thank you. I thought USB-C was only for power? Is it the reverse, Micro-USB for power and USB-C for serial communication?

What ports should be represented under Ports in Hardware Manager on Windows 11? 

Thanks for clarifications!

sporkyduck, 2024-04-11:
Ok, I see. Thank you very much for a quick response, appreciated!

closed 2024-04-11 by sporkyduck