Xinyuan-LilyGO/LilyGo-T-SIM7080G #61: Ways to check power status ?

dannybackx, 2024-04-22:
Hey,
I bought this board to be able to send warnings even if the house network is down, or the power.
So can the board measure its supply voltage by itself ? Or do I need external circuitry ?
Same question for the battery : can I measure its voltage level ?

Thanks

lewisxhe, 2024-04-29:
This board has a PMU that can monitor the battery voltage. If it is an external voltage, you need to build your own circuit to measure it.

dannybackx, 2024-04-29:
Meanwhile I have been experimenting and found out it can measure quite some stuff so looks like I can both detect when the battery is low but also when external power goes out and comes back : 
```
I (18:38:40.891) power: Battery voltage 4199, percentage 86
I (18:38:40.898) power: Power: isCharging: YES
I (18:38:40.903) power: Power: isDischarge: NO
I (18:38:40.908) power: Power: isVbusIn: YES
I (18:38:40.913) power: Power: getBattVoltage: 4199 mV
I (18:38:40.919) power: Power: getVbusVoltage: 4969 mV
I (18:38:40.925) power: Power: getSystemVoltage: 4940 mV
```

closed 2024-05-18 by dannybackx