Xinyuan-LilyGO/LilyGo-T-SIM7080G #62: / LilyGo T-SIM7080G board vs humidity ?

mathieucarbou, 2024-04-25:
Hello,

I am testing the lilygo board outside since a few days. There is a lot of humidity outside, the lilygo is in a waterproof box (but the box has a hole for the wires on the side, so the air can enter but not the rain).

I am wondering what's this on the board ? it was not there before. It feels like the humidity stats damaging the board ?

[image]

The connected usb-c device is this solar panel : https://fr.aliexpress.com/item/1005005509831452.html

* Tension de fonctionnement: 5 ~ 5.5V 
* Courant de travail: 0.8A ~ 1A(Max.)

lewisxhe, 2024-04-26:
Obviously it is affected by moisture. You can remove the damaged original parts first, then power on and test whether they can work.

mathieucarbou, 2024-04-26:
> Obviously it is affected by moisture. You can remove the damaged original parts first, then power on and test whether they can work.

I don't have the tools for such precise work at home. Right now, I can power the board with USB-C, solar panel connector does no work, and I can power through internal battery.

But I found that when I have a USB-C solar panel connected and internal battery, when the solar panel is not able to produce enough current, the device shuts down instead of switching to the battery. 

I do not have this behaviour with the second board.

I have ordered a spray for electronic boards... 

But what I don't understand is why these boards come without any varnish, silicon or resin to protect the components from humidity / moisture ?

Also, is there a solution I could do to cleanup without having to remove the components ?

How I could check if some are damaged ?

mathieucarbou, 2024-04-29:
@lewisxhe : do you know what kind of product I could use to cleanup and remove this blue-green mixture ?

closed 2024-06-13 by github-actions[bot]