Xinyuan-LilyGO/LilyGo-T-SIM7080G #64: Bringing the board back online when re-charging a depleted 18650 battery

Peer77, 2024-04-30:
Hi Lewis,

when comparing the LilyGo SIM7080G board with the SIM800L, I observed the charging / repowering of SIM7080G is proceeding differently when the 18650 battery is discharged completely and the device went off. Please see the following scenario.

 **Behaviour of SIM800L:**

- When an empty 18650 battery is inserted (or the board discharged the 18650 until it switched off) and 5V is connected to 5V-IN pin, the battery was charging and the board came back online when the 18650 voltage was sufficient(it tried in between booting up until it succeeded)

-> Is there a way to apply this behaviour to the SIM7080G as well?

**Current behaviour of SIM7080G:**

- When an empty / depleted battery is inserted and 5V is fed to the pad (like when the power grid comes back after power interruption) of the removed diode (as mentioned in #49), it tried to boot up (but failed of course as expected). It went off and the battery was then charging normally (indicated by blue charging LED), but it charged only to 3.63V. The charging LED was then flashing at 1Hz, but the board did not come back online unfortunately. I had to remove the auxiliary
5V line (going to diode pad) and reconnect it to make it booting up again.

-> Is there a way to change that so it comes back online automatically? Or is it possible to set a higher threshold by using PMU.setLowBatShutdownThreshold() to shut down the board earlier to have sufficient power for booting up when the power grid comes back? The battery I used for this test was discharged by the board to 2.4V. 

Many thanks! and best regards, 

Peer

lewisxhe, 2024-05-02:
If it is lower than the PMU operating voltage, the PMU must be turned on manually. You can set the low voltage alarm to trigger the PMU interrupt to shut down in advance or enter sleep mode to save power. I have uploaded a simple test example, I hope it will be helpful to you.

Peer77, 2024-05-13:
Hi Lewis,
thank you very much, I will check and report back...

Peer77, 2024-05-24:
Hi Lewis,

thank you for the example code. Your code brought some inspiration. So I have written a small function to simply check against the return value of PMU.isVbusIn(). If powergrid goes down it returns false, and after a 1 minute I then call PMU.shutdown();. Feeding back the PSU / 5V to the pad where the diode was mounted, the device comes back online as normal. Works fine! Just to confirm, the board is really switched off after calling PMU.shutdown(), right?

Thanks!

Peer77, 2024-06-04:
Resolved, closing.

closed 2024-06-04 by Peer77