Xinyuan-LilyGO/LilyGo-T-SIM7080G #66: using modem on one core, and another I2C device on another

Bfreak, 2024-05-06:
Hi. I am trying to run an MQTT script on one core, and read data from another i2c device on the other core, using freertos to do so. 

When I try to do this, I get 'failed to initialize power' from the modem.

The MQTT tasks has default code:
```
void MQTT_task_setup(void *pvParameters) {

    Serial.begin(115200);

    //Start while waiting for Serial monitoring
    while (!Serial);

    vTaskDelay(3000);

    Serial.println();

    if (!PMU.begin(Wire, AXP2101_SLAVE_ADDRESS, I2C_SDA, I2C_SCL)) {
        Serial.println("Failed to initialize power.....");
        while (1) {
            vTaskDelay(5000);
        }
    }
```

and the other i2c has code

```
void hall_task_setup(void *Parameter) {
  Serial.println("Starting Hall_get setup");

  Wire.begin(13, 21); 

  if (! sensor.begin_I2C()) {    
  //if (! sensor.begin_SPI(MLX90393_CS)) {  // hardware SPI mode
    Serial.println("No sensor found ... check your wiring?");
    while (1) { vTaskDelay(10); }
  }
  Serial.println("Found a MLX90393 sensor");
```

Both can run independantly using freertos. Any suggestions? Thanks.

lewisxhe, 2024-05-07:
There is only one Wire. You initialized it twice. Of course it cannot run. You need to change one to Wire1.

lewisxhe, 2024-05-07:
According to the question you raised, https://github.com/Xinyuan-LilyGO/LilyGo-T-SIM7080G/issues/65, you directly pass the PMU initialization to Wire1, so that your sensors will not conflict

Bfreak, 2024-05-07:
thanks @lewisxhe but I have a new problem: I currently can only upload script when I connect via usb holding boot button , but get nothing in terminal. When I try and connect without boot I get either `*** [upload] Error 1` or `error 13: could not open port` when I try and open any connection. Any suggestion?

closed 2024-06-23 by github-actions[bot]