Xinyuan-LilyGO/LilyGo-T-SIM7080G #67: missing example: using camera with sim7080.

Bfreak, 2024-05-14:
Thanks Lilygo for the excellent product, thanks to @lewisxhe for well written examples for MQTT.

However, I think many who purchase the board are expecting to use the camera module with FTP or HTTP.

Can I please suggest an example of posting an image to FTP or HTTP is created? I attempted to create one myself, but there is also no example of filesystem management for SIMCOM7080 in examples, and also it seems to integration for filesystem management in TinyGSM library.

lewisxhe, 2024-05-23:
I don't have time to write other examples now, but here are how to use FTP and FS 
https://github.com/Xinyuan-LilyGO/LilyGo-T-SIM7080G/blob/master/datasheet/SIM7070_SIM7080_SIM7090%20Series_FS_Application%20Note_V1.02.pdf
https://github.com/Xinyuan-LilyGO/LilyGo-T-SIM7080G/blob/master/datasheet/SIM7070_SIM7080_SIM7090%20Series_FTP(S)_Application%20Note_V1.02.pdf

Bfreak, 2024-05-23:
Thanks Lewis,

however, this doesn't include the process of streaming a file over UART
from from the board. Does any other lilygo example cover this?

Thanks

lewisxhe, 2024-05-24:
This is the communication command of SIM7080G. You can interact with SIM7080 through ATDebug.

Bfreak, 2024-05-24:
Hi,

That's not the problem. The problem is writing data to modem over UART
after AT+CFSWFILE, this can not be done with at debug.

Bfreak, 2024-05-24:
we attempt to upload an image to the modem to send to FTP server with the
following code

void sendFileToModem(File file, const char* filename) {
>     modem.sendAT("+CFSTERM");
>     modem.waitResponse();
>     modem.sendAT("+CFSINIT");
>     modem.waitResponse();
>     String command = "+CFSWFILE=0,\"" + String(filename) + "\",0," +
> String(file.size()) + ",10000";
>     Serial.println(command);
>     modem.sendAT(command.c_str());
>     modem.waitResponse(30000UL, "DOWNLOAD");
>     modem.stream.write(file.read());
>     modem.waitResponse(30000UL);
>     modem.sendAT("+CFSGFRS?");
>     modem.waitResponse();
> }

output after download is always error

AT
> OK
> AT+CFSTERM
> AT+CFSTERM
> OK
> AT+CFSINIT
> AT+CFSINIT
> OK
> +CFSWFILE=0,"image.png",0,2355,10000
> AT+CFSWFILE=0,"image.png",0,2355,10000
> AT+CFSWFILE=0,"image.png",0,2355,10000
> DOWNLOAD�

> ERROR
> AT+CFSGFRS?
> AT+CFSGFRS?
> +CFSGFRS: 5851136

> OK

Any suggestion here is appreciated. Thank you.

closed 2024-07-08 by github-actions[bot]