Xinyuan-LilyGO/LilyGo-T-SIM7080G #70: UART communication

manzajaro, 2024-05-22:
Hi, I'm testing the UART communication between this device and an arduino, but I can't get it to work.

On the pin diagram it says TXD and RXD are numbers 43 and 44. but I can't get it to work, can you help me?

`void setup() {
  Serial.begin(115200);
  while (!Serial);
  
  Serial1.begin(9600, SERIAL_8N1, GPIO_NUM_43, GPIO_NUM_44);

  delay(1500);

  String message = "";
  long timeCheck = millis();

  while (message.equals("") && (millis() - timeCheck) < 5000) {
    if (Serial1.available() > 0) {  // Verifica si hay datos disponibles para leer
      message = Serial1.readString();  // Lee el mensaje recibido
      Serial.print("Mensaje recibido: ");
      Serial.println(message);  // Imprime el mensaje recibido
    }
  }
}`

Thanks

lewisxhe, 2024-05-23:
If you need to use a modem, then you'd better change Serial1 to Serial2, and you must set CDC On boot in the Arduino settings to 1, otherwise it will not work

manzajaro, 2024-05-23:
Thanks lewisxhe.

My idea is to run the project on battery power, as I read in the following comment for it to work without USB power, that field must be set to ‘disabled’. Will it work?

https://github.com/Xinyuan-LilyGO/LilyGo-T-SIM7080G/issues/38#issuecomment-1891335599

lewisxhe, 2024-05-24:
Yes, when you use the battery, just don't initialize Serial. Turn it on when you need to print and debug.

closed 2024-05-29 by manzajaro