Xinyuan-LilyGO/LilyGo-T-SIM7080G #74: I do not get data through the serial console

famphuelva, 2024-06-24:
#include <Arduino.h>
#define XPOWERS_CHIP_AXP2101
#include "XPowersLib.h"
#include "utilities.h"

XPowersPMU PMU;

// See all AT commands, if wanted
#define DUMP_AT_COMMANDS

#define TINY_GSM_RX_BUFFER 1024

#define TINY_GSM_MODEM_SIM7080
#include <TinyGsmClient.h>
#include "utilities.h"

#ifdef DUMP_AT_COMMANDS
#include <StreamDebugger.h>
StreamDebugger debugger(Serial1, Serial);
TinyGsm modem(debugger);
#else
TinyGsm modem(SerialAT);
#endif

const char *register_info[] = {
    "Not registered, MT is not currently searching an operator to register to.The GPRS service is disabled, the UE is allowed to attach for GPRS if requested by the user.",
    "Registered, home network.",
    "Not registered, but MT is currently trying to attach or searching an operator to register to. The GPRS service is enabled, but an allowable PLMN is currently not available. The UE will start a GPRS attach as soon as an allowable PLMN is available.",
    "Registration denied, The GPRS service is disabled, the UE is not allowed to attach for GPRS if it is requested by the user.",
    "Unknown.",
    "Registered, roaming.",
};

enum {
    MODEM_CATM = 1,
    MODEM_NB_IOT,
    MODEM_CATM_NBIOT,
};

#define randMax 35
#define randMin 18

// Credenciales de la red
const char apn[] = "*******************************************";
const char gprsUser[] = "************************************";
const char gprsPass[] = "*************************************";

// cservidor mosqitto
const char server[] = "*******************************************";
const int port = 1883;
char buffer[1024] = {0};
char username[] = "**********************************************";
char password[] = "**************************************************";
char clientID[] = "temperatura";
int data_channel = 0;

// Número máximo de intentos de reconexión
const int maxReconnectAttempts = 10;

bool isConnect() {
    modem.sendAT("+SMSTATE?");
    if (modem.waitResponse("+SMSTATE: ")) {
        String res = modem.stream.readStringUntil('\r');
        return res.toInt();
    }
    return false;
}

void restartModem() {
    digitalWrite(BOARD_MODEM_PWR_PIN, LOW);
    delay(100);
    digitalWrite(BOARD_MODEM_PWR_PIN, HIGH);
    delay(1000);
    digitalWrite(BOARD_MODEM_PWR_PIN, LOW);
    delay(3000); // Esperar a que el módem se reinicie completamente
}

void reconnectMQTT() {
    int attempts = 0;
    while (attempts < maxReconnectAttempts) {
        modem.sendAT("+SMCONN");
        if (modem.waitResponse(30000) == 1) {
            Serial.println("MQTT Conectado!");
            return;
        }
        Serial.println("Conexión fallida, reconectando con MQTT ...");
        delay(1000);
        attempts++;
    }
    Serial.println("Número máximo de intentos de reconexión alcanzado, reiniciando el módem...");
    restartModem();
}

void setup() {
    Serial.begin(115200);
    while (!Serial);

    delay(3000);

    Serial.println();

    if (!PMU.begin(Wire, AXP2101_SLAVE_ADDRESS, I2C_SDA, I2C_SCL)) {
        Serial.println("Failed to initialize power.....");
        while (1) {
            delay(5000);
        }
    }

    PMU.setDC3Voltage(3000);
    PMU.enableDC3();
    PMU.setBLDO2Voltage(3300);
    PMU.enableBLDO2();
    PMU.disableTSPinMeasure();

    Serial1.begin(115200, SERIAL_8N1, BOARD_MODEM_RXD_PIN, BOARD_MODEM_TXD_PIN);

    pinMode(BOARD_MODEM_PWR_PIN, OUTPUT);
    pinMode(BOARD_MODEM_DTR_PIN, OUTPUT);
    pinMode(BOARD_MODEM_RI_PIN, INPUT);

    int retry = 0;
    while (!modem.testAT(1000)) {
        Serial.print(".");
        if (retry++ > 6) {
            restartModem();
            retry = 0;
            Serial.println("Retry start modem.");
        }
    }
    Serial.println();
    Serial.print("Modem inicializado!");

    if (modem.getSimStatus() != SIM_READY) {
        Serial.println("Tarjeta SIM no insertada!!!");
        return;
    }

    modem.setNetworkMode(2);
    modem.setPreferredMode(MODEM_NB_IOT);

    uint8_t s;
    do {
        s = modem.getRegistrationStatus();
        if (s != REG_OK_HOME && s != REG_OK_ROAMING) {
            Serial.print(".");
            delay(1000);
        }
    } while (s != REG_OK_HOME && s != REG_OK_ROAMING);

    Serial.println();
    Serial.print("Informacion de la Red registrada:");
    Serial.println(register_info[s]);

    if (!modem.isGprsConnected()) {
        modem.sendAT("+CNACT=0,1");
        if (modem.waitResponse() != 1) {
            Serial.println("Portadora de la red fallida!");
            return;
        }
    }

    Serial.print("GPRS estado:");
    Serial.println(modem.isGprsConnected() ? "conectado" : "no conectado");

    modem.sendAT("+SMDISC");
    modem.waitResponse();

    snprintf(buffer, 1024, "+SMCONF=\"URL\",\"%s\",%d", server, port);
    modem.sendAT(buffer);
    if (modem.waitResponse() != 1) {
        return;
    }
    snprintf(buffer, 1024, "+SMCONF=\"USERNAME\",\"%s\"", username);
    modem.sendAT(buffer);
    if (modem.waitResponse() != 1) {
        return;
    }
    snprintf(buffer, 1024, "+SMCONF=\"PASSWORD\",\"%s\"", password);
    modem.sendAT(buffer);
    if (modem.waitResponse() != 1) {
        return;
    }
    snprintf(buffer, 1024, "+SMCONF=\"CLIENTID\",\"%s\"", clientID);
    modem.sendAT(buffer);
    if (modem.waitResponse() != 1) {
        return;
    }

    reconnectMQTT();
    randomSeed(esp_random());
}

void loop() {
    if (!isConnect()) {
        Serial.println("MQTT Cliente desconectado!");
        reconnectMQTT();
        delay(1000);
        return;
    }

    Serial.println();
    String payload = "";
    int temp = rand() % (randMax - randMin) + randMin;
    payload.concat(temp);
    snprintf(buffer, 1024, "+SMPUB=\"%s/f/%s\",%d,1,1", username, clientID, payload.length());

    modem.sendAT(buffer);
    if (modem.waitResponse(">") == 1) {
        modem.stream.write(payload.c_str(), payload.length());
        Serial.print("Try publish payload: ");
        Serial.println(payload);
        if (modem.waitResponse(3000)) {
            Serial.println("Send Packet success!");
        } else {
            Serial.println("Send Packet failed!");
        }
    }

    delay(300000);
}

This code works perfectly, it sends the data to the mqtt server, but it does not show me information on the PC console, is the board broken? I have changed the code for a simple one that the console writes and it doesn't show anything either. Is the board bad?

famphuelva, 2024-07-07:
The serial does not seem to work well, with a simple code on a new board this code does not show anything:
`void setup(){
  Serial.begin(9600);
}

void loop(){

  Serial.println("sgfdhsgfhs");
  delay(1000);
}`

lewisxhe, 2024-07-08:
[image]

famphuelva, 2024-07-08:
thank you very much, solved.

closed 2024-08-22 by github-actions[bot]