Xinyuan-LilyGO/LilyGo-T-SIM7080G #76: How to use eSIM?

andreiandreiandreiandreiandrei, 2024-07-04:
Hi,

I didn't find any guidance on how to set up an eSIM. In the file SIM7080G_Hardware_Design.pdf at page 33 there is a single mention in the note: "There is an eSIM card slot reserved inside the SIM7080G module. The specific supported protocols
and card operations need to be provided by the card manufacturer". Well as the eSIM card is not removable, which manufacturer other than Simcom should provide guidance.

For a phone, the service provider provides a link to download a file that is written somewhere in the eSIM.  In our case, we should obtain the file from the provider, and then what? how to write it in the eSIM?

lewisxhe, 2024-07-08:
Are you mistaken? SIM7080G does not support eSIM function

andreiandreiandreiandreiandrei, 2024-07-08:
Hi,

As I mentioned, the documentation stated that it does.

Plus here at simcom page in product selector:

[image]

plus on simcom page at product description at interfaces says: " Embedded SIM (optional) ". 
So that could be a clue that not all the chips have eSIM, but some could have.

Now, if you have more info regarding this feature: which chips have. And the ones that have how could be provisioned

lewisxhe, 2024-07-11:
That is a customized type. LilyGo's SIM7080 does not have a built-in eSIM.

andreiandreiandreiandreiandrei, 2024-07-11:
Thank you for your answer

closed 2024-07-11 by andreiandreiandreiandreiandrei