Xinyuan-LilyGO/LilyGo-T-SIM7080G #78: PlatformIO Serial monitor doesn't give correct output

27rabbitlt, 2024-07-29:
Hi Lewis~

We are previously using SIM7000G and noticed SIM7080G seems better, so I tried SIM7080G with the same code.

It compiles, uploads correctly but when I tried to monitor Serial, the output is something like:

```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x8 (TG1WDT_SYS_RST),boot:0x28 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40378832
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbe4
load:0x403cc700,len:0x2a38
entry 0x403c98d4
```

I found a post on forum but the I don't know how to apply that solution to platformIO's Serial monitor: https://community.platformio.org/t/esp32-s3-fh4n2-weact-studio-mini-wont-run-with-platformio-settings-several-issues/40442/3

The configuration file platformio.ini is copied from sample code:

```
[env:eride_tracker]
platform = espressif32@6.3.0
board = esp32s3box
framework = arduino
monitor_speed = 115200
lib_deps = 
	knolleary/pubsubclient@^2.8
	vshymanskyy/TinyGSM@^0.11.5
	vshymanskyy/StreamDebugger@^1.0.1
	bblanchon/ArduinoJson@^6.18.3
	lewisxhe/XPowersLib@^0.2.1
	paulstoffregen/Time@^1.6.1
upload_speed = 921600
monitor_filters = default, log2file

build_flags =
    -DBOARD_HAS_PSRAM       

    ; Enable UARDUINO_USB_CDC_ON_BOOT will start printing and wait for terminal access during startup
    -DARDUINO_USB_CDC_ON_BOOT=1

    ; Enable UARDUINO_USB_CDC_ON_BOOT will turn off printing and will not block when using the battery
    ; -UDARDUINO_USB_CDC_ON_BOOT

    -DTINY_GSM_MODEM_SIM7080
    -DTINY_GSM_RX_BUFFER=1024
    -DCAYENNE_PRINT=Serial
    -DCONFIG_BT_BLE_50_FEATURES_SUPPORTED

board_build.partitions = huge_app.csv  

[platformio]
description = eride sim7000G prototyping
src_dir = src
```

Then I tried a example code and found that Serial output is correct. So it couldn't  be hardware problem.

Any hints or suggestions?

lewisxhe, 2024-08-22:
USB CDC needs to be set to 1

romwam, 2024-10-01:
I (successfully) use following defines in the platform.ini at "build_flags":
	-DARDUINO_USB_MODE=1
	-DARDUINO_USB_CDC_ON_BOOT=1

closed 2024-10-04 by lewisxhe