Xinyuan-LilyGO/LilyGo-T-SIM7080G #79: External power supply with newer version of T-SIM7080?

romwam, 2024-09-12:
I have followed successfully the directions to power the device by external voltage as given in issue
https://github.com/Xinyuan-LilyGO/LilyGo-T-SIM7080G/issues/30 with my first samples of T-SIM7080 last year.

Unforunately on my newer board the diode is not on the board any more and connecting +5V to the indicated diode pad does not power up the board any more.

Please indicate how to power the T-SIM7080 using external 5V voltage for the newer boards where no diode is placed any more.

thank you,

Roman

lewisxhe, 2024-09-13:
The PCB is exactly the same, just removed the diodes and nothing else changed!

lewisxhe, 2024-09-13:
Yes

romwam, 2024-09-13:
Thank you very much for your quick response!

closed 2024-10-04 by lewisxhe