Xinyuan-LilyGO/LilyGo-T-SIM7080G #81: Example "MinimalModemNBIOTExample" not working, compilation error

rFlightFPV, 2024-10-03:
I'm trying to compile the example **MinimalModemNBIOTExample** and have set my APN in line 68, the only change to the code. I get a compilation error in Arduino IDE 2.3.3 as follows:

_error: 'RegStatus' was not declared in this scope; did you mean 'SimStatus'?_

Thank you for your help!

rFlightFPV, 2024-10-03:
SOLVED: Install libraries as per instructions ("Copy all the folders in the LilyGo-T-SIM7080G/lib directory to <C:\Users\UserName\Documents\Arduino\libraries>. ").

rFlightFPV, 2024-10-13:
@lewisxhe could you please try to replicate the following? If I copy all the libraries in the "lib" folder of this repository to my libraries folder the "MinimalModemNBIOTExample.ino" sketch complies fine. However, if I update my libraries I get the error _error: 'RegStatus' was not declared in this scope; did you mean 'SimStatus'?_
I pinned it down to the TinyGSM library. The sketch compiles with TinyGSM 0.11.5, with the current 0.12.0 it does not.

As far as I saw within the TinyGSM library it now has added functionality for the SIM7080 modem. It would imo be great if the sketch could be updated accordingly, many thanks!

lewisxhe, 2024-10-14:
Already updated, but not sure if other features will be affected.

closed 2024-11-27 by github-actions[bot]

MikeyJohnston, 2024-12-13:
I've been working towards optimizing one of these lilygo-t-sim7080 boards and I have it publishing GPS data to a server through HTTP connection/cellular about every 30 seconds. Is this about as fast as I can get this to update to the server? Or do you think I can get it faster?