Xinyuan-LilyGO/LilyGo-T-SIM7080G #83: LilyGo-T-SIM7080G -  examples/ATDebug.ino could not start the modem

wsiripun, 2024-10-06:
Hello,

Problem:  examples/ATDebug.ino could not start the modem.
Hardware Version:   "T-SIM7080G V1.0" (the label is near the battery holder)
IDE:   Visual Studio Code with PlatformIO 6.1.16

The examples/ATDebug.ino could not start the modem.  The serial monitor kept looping showing this message:

 ***********************************************************
 Failed to connect to the modem! Check the baud and try again.     <--- Kept looping on this message
***********************************************************

The message is from line 109 of ATDebug.ino.  I have tried changing the baud rate to 9600, but it did not help.

(Note: ATDebug.ino was not modified in anyway.  Only platformio.ini was modified to make it compiled and loaded.)

Please help.  Thanks.

lewisxhe, 2024-10-07:
If you have a multimeter, you can check whether the voltage at these two points is 3.4V
[image]

lewisxhe, 2024-10-08:
If voltage is present, the modem's red status light should be on and it should be able to receive communications, right?

wsiripun, 2024-10-08:
I could not make ATDebug work again, but MinimalModemNBIOTExample.ino worked (using Hologram SIM).When MinimalModemNBIOTExample worked, the two red LED's near the GPS antenna showed one solid red and one blinking red.  The blue LED between SIM and TF card was off. The voltage was 3.054 as before.

When ATDebug did not work, the two red LED's were off.  The blue LED showed solid blue. The voltage was zero.

lewisxhe, 2024-10-09:
I found that the problem is that PMU turns on DC3 by default, causing DC3 to be in protection state at the moment of startup. Please update the example to turn off DC3 after initializing PMU, and then delay for a while before starting again.

closed 2024-11-23 by github-actions[bot]