Xinyuan-LilyGO/LilyGo-T-SIM7080G #88: Battery not supplying power

sporkyduck, 2024-11-21:
Hi!

One of several SIM70G:s I have has stopped supplying power when the mechanical battery switch is turned on. When switch is off, battery is not detected with pmu.isBatteryConnect(), which is normal. When switch is on, battery is detected with seems to be correct level percent  from pmu.getBatteryPercent(), ~100% at 4.16V. This is all while USB is connected. When I detach USB, power turns off. 

Pressing PWR has once resultated in a very short startup try (GPS and modem LEDS came on) and but then off.

Is this a hardware or software issue, and how can I check?

Thanks!

lewisxhe, 2024-11-21:
You said you have several machines, and one of them has this problem?

lewisxhe, 2024-11-21:
Please turn on the switch, then connect the battery and test whether there is voltage here.
[image]

sponsoredlinks, 2024-11-25:
Note we have the same issue with our T-SIM7080G v1.0

Was working the other day but strangely - with a battery in it would not power up until USB-C was inserted - would then run fine

Now it doesn't work with either battery or battery  +USB

Voltage at the pin indicated in the picture is 4.050V

Is there anything else we can do to debug?

lewisxhe, 2024-11-25:
If there is voltage here, then the power supply has been connected to the PMU. Will pressing the PWR button not turn on the computer?

sponsoredlinks, 2024-11-25:
No, unfortunately not. We had just loaded the MinimalPowers reporting
sketch and then tried the MinimalPowersCurrent.

We then reverted to the MinimalPowersExample. Then it didnt turn on two
days later

We have tried pressing and holding each of the buttons

The issue with it being unable to power on with battery only (until you
plug usb, but then staying on even after usb detach) is possibly related?

lewisxhe, 2024-11-25:
Using [MinimalPowersCurrentExample](https://github.com/Xinyuan-LilyGO/LilyGo-T-SIM7080G/blob/master/examples/MinimalPowersCurrentExample/MinimalPowersCurrentExample.ino) , press the PWR button. Does it print anything?

sponsoredlinks, 2024-11-25:
I dont believe we can get it to power on to load the example? Will try
tomorrow and report back - thank you for the help

sponsoredlinks, 2024-11-26:
Thanks Lewis it turned out you were right

The examples have an endless loop at the start that waits for serial input to be connected

So if you just turn it on on battery, basically it waits endlessly for a serial connection and doesn't provide any signs of life until serial is plugged in -> 
https://github.com/Xinyuan-LilyGO/LilyGo-T-SIM7080G/blob/master/examples/MinimalPowersExample/MinimalPowersExample.ino#L30

I may have missed it in the documentation but if not already there it might be worth making this obvious (or flashing the led perhaps  :)

thanks

lewisxhe, 2024-11-26:
OK, then it is actually started, and CDC is waiting for serial access

closed 2025-01-09 by github-actions[bot]