Xinyuan-LilyGO/LilyGo-T-SIM7080G #90: SIM7080G Unable to Register on Network, +CSQ Returns 99

pausalv, 2024-12-10:
### **Description**  
I am experiencing an issue where the SIM7080G module fails to register on the network. When I send the `AT+CSQ` command, it consistently returns `+CSQ: 99`, indicating "unknown or undetectable signal strength."  

This issue occurs on one specific board, while the same software and SIM card work fine on other boards using the same module.  

---

### **Expected Behavior**  
The module should return a valid signal strength and attempt to register on the network.  

---

### **Actual Behavior**  
- The module fails to detect signal (`+CSQ: 99`).  
- Network registration status (`AT+CREG?` or `AT+CEREG?`) alternates between "Not registered, searching" and "Not registered."  

---

### **Troubleshooting Steps Taken**  
1. Verified the SIM card is active and functional.  
2. Tested the same software and SIM on other boards, which work fine.  
3. Confirmed proper antenna connection and board power supply.  
4. Sent various AT commands to diagnose the issue:  
   - `AT+CSQ`: `+CSQ: 99`  
   - `AT+CREG?`: Alternates between `0,2` and `0,0`.  

---

### **Questions/Request**  
1. Could this be a hardware defect in the specific board?  
2. Are there additional diagnostics or configurations I should try?  
3. Any known compatibility issues or common causes for `+CSQ: 99`?  

Thank you for your assistance!

lewisxhe, 2024-12-11:
You said you have two boards. You can use the following command to compare the return values ​​of the two boards to see if they are the same;
```bash
AT+CNMP?
AT+CMNB?
AT+CBANDCFG?
```

pausalv, 2024-12-11:
Thank you for reporting the issue. Based on your response, it seems the problem was related to an incomplete configuration of the NB-IoT bands on the problematic board.

**Resolution:**  

The issue was resolved by properly configuring the NB-IoT bands on the problematic board.  

### Steps Taken:  
1. The following command was used to update the NB-IoT band configuration:  
   ```
   AT+CBANDCFG="NB-IOT",1,2,3,4,5,8,12,13,18,19,20,25,26,28,66,71,85
   ```  

2. After applying this configuration, the module successfully connected to the network.  

### Recommendation:  
If experiencing similar issues, check the current band configuration with `AT+CBANDCFG?` and ensure it includes all necessary bands.

closed 2024-12-11 by pausalv