Xinyuan-LilyGO/LilyGo-T-SIM7080G #91: what is the smallest time interval this device can report GPS data to a server

MikeyJohnston, 2024-12-13:
I've been working towards optimizing one of these lilygo-t-sim7080 boards and I have it publishing GPS data to a server through HTTP connection/cellular about every 30 seconds. Is this about as fast as I can get this to update to the server? Or do you think I can get it faster?
my goal is 5 seconds

lewisxhe, 2024-12-16:
This depends on the interval of obtaining positioning and the time used for transmission. SIM7080G cannot run GPS at the same time when the network is turned on, so this greatly increases the GPS transmission interval. You need to test it yourself. Different networks and different usage environments are different. I can't give you this data.

lewisxhe, 2024-12-17:
Try using the following command to shut down instead of shutting down the whole power
[image]

MikeyJohnston, 2024-12-18:
Thank you for the recommendation!! I have tried to incorporate putting it into lowest power mode for GPS but am unable to push to HTTP with the power on. I think I may have better luck with the SIM7000. I'm not going to count this device out but am going to get some testing in with the SIM7000 and re-evaluate after that.

closed 2025-02-01 by github-actions[bot]