Xinyuan-LilyGO/LilyGo-T-SIM7080G #93: In MinimalModemNBIOTExample, testAT never succeeds

eduard-firewave, 2024-12-30:
I'm running the MinimalModemNBIOTExample code and testAT method never succeeds. I see that "AT" is printed all the time in the SerialMonitor. I think it worked once on this device, but now it never works. Is there a way to reset the SIM module somehow?

Here is the sample output:
AT
�.AT
.AT
.AT
.AT
.AT
.AT
.AT
.Retry start modem .
AT
.AT
����.AT
.AT
.AT
.AT
��.AT
.AT
.Retry start modem .

lewisxhe, 2024-12-31:
1. Has the code been modified?
2. Is the red light next to the modem on?

lewisxhe, 2024-12-31:
Have you tried pressing and holding the ESP32 BOOT button and then plugging in the USB? Please note that there is a SIM BOOT button next to it, so don’t press the wrong one.

eduard-firewave, 2024-12-31:
1. The code hasn't been modified (besides the apn string), but previously another code was uploaded.
2. Yes, the red light is on and the other one beside it is blinking.

Thanks!

lewisxhe, 2024-12-31:
Have you changed the baud rate of the modem?

If the red light is flashing, the modem is normal. You can insert the micro-USB into the USB port next to the modem, install the SIM7080 driver, then select the AT port, change the baud rate to 115200, and then send AT to see if it returns OK

https://github.com/Xinyuan-LilyGO/LilyGo-T-SIM7080G/blob/master/docs/sim7080_update_firmware.md

lewisxhe, 2024-12-31:
The link is just to show you how to update the driver, please do not update the firmware

eduard-firewave, 2024-12-31:
Is there similar guide for mac?

lewisxhe, 2024-12-31:
Can only be used on Windows, no Mac

eduard-firewave, 2024-12-31:
will try on windows machine and will update... thanks for your support

eduard-firewave, 2024-12-31:
and now the update tool can not update again for some reason:

[image]

eduard-firewave, 2024-12-31:
It does seem to work from Arduino IDE though:
[image]

But still no success with the code from the example. It still doesn't get any response for AT command.

eduard-firewave, 2024-12-31:
I updated the baud rate using Arduino IDE Serial Monitor and the "AT+IPR=115200" command, and it fixed the issue. Thanks for you support!

I f I can, a follow up question:
Is it possible to increase the baud rate of the AT port?

Thanks!

closed 2025-01-01 by eduard-firewave