Xinyuan-LilyGO/LilyGo-T-SIM7080G #94: SD card example throws errors randomly

eduard-firewave, 2025-01-01:
Hi, I'm running the SD Card example and I get the errors on each run. The errors seem to appear in random places - not consistently. Here is the sample output:

SD_MMC Card Type: SDHC
SD_MMC Card Size: 29819MB
Listing directory: /
  FILE: datalog.txt  SIZE: 15690823
  FILE: tmp.raw  SIZE: 353280
  FILE: currentVersion  SIZE: 1
  DIR : .Spotlight-V100
  DIR : .fseventsd
  FILE: ._datalog.txt  SIZE: 4096
  FILE: ._tmp.raw  SIZE: 4096
  FILE: test.txt  SIZE: 0
  FILE: foo.txt  SIZE: 13
Creating Dir: /mydir
**E (4751) sdmmc_req: sdmmc_host_wait_for_event returned 0x107
E (4751) diskio_sdmmc: sdmmc_write_blocks failed (263)
E (4752) sdmmc_cmd: sdmmc_read_sectors_dma: sdmmc_send_cmd returned 0x109
E (4757) diskio_sdmmc: sdmmc_read_blocks failed (265)**
mkdir failed
Listing directory: /
**E (4763) sdmmc_req: handle_idle_state_events unhandled: 00000008 00000002**
  FILE: datalog.txt  SIZE: 15690823
  FILE: tmp.raw  SIZE: 353280
  FILE: currentVersion  SIZE: 1
  DIR : .Spotlight-V100
  DIR : .fseventsd
  FILE: ._datalog.txt  SIZE: 4096
  FILE: ._tmp.raw  SIZE: 4096
  FILE: test.txt  SIZE: 0
  FILE: foo.txt  SIZE: 13
Removing Dir: /mydir
**[  4878][E][vfs_api.cpp:250] rmdir(): /mydir does not exists or is a file**
rmdir failed
Listing directory: /
  FILE: datalog.txt  SIZE: 15690823
  FILE: tmp.raw  SIZE: 353280
  FILE: currentVersion  SIZE: 1
  DIR : .Spotlight-V100
Listing directory: /.Spotlight-V100
  DIR : Store-V2
Listing directory: /.Spotlight-V100/Store-V2
  DIR : 42E83897-C3B2-4FA8-954A-0313405B4665
  FILE: VolumeConfiguration.plist  SIZE: 4506
  DIR : .fseventsd
Listing directory: /.fseventsd
  FILE: fseventsd-uuid  SIZE: 36
  FILE: ._datalog.txt  SIZE: 4096
  FILE: ._tmp.raw  SIZE: 4096
  FILE: test.txt  SIZE: 0
  FILE: foo.txt  SIZE: 13
Writing file: /hello.txt
File written
Appending to file: /hello.txt
Message appended
Reading file: /hello.txt
Read from file: Hello World!
Deleting file: /foo.txt
File deleted
Renaming file /hello.txt to /foo.txt
File renamed
Reading file: /foo.txt
Read from file: Hello World!
0 bytes read for 0 ms
**E (4849) sdmmc_req: handle_idle_state_events unhandled: 00000042 00000000
E (4854) sdmmc_req: handle_idle_state_events unhandled: 00000042 00000000
E (4861) sdmmc_req: handle_idle_state_events unhandled: 00000040 00000000
E (4865) sdmmc_req: handle_idle_state_events unhandled: 00000042 00000000
E (4867) sdmmc_req: handle_idle_state_events unhandled: 00000042 00000000
E (5871) sdmmc_req: sdmmc_host_wait_for_event returned 0x107
E (5871) diskio_sdmmc: sdmmc_write_blocks failed (263)
E (6872) sdmmc_req: sdmmc_host_wait_for_event returned 0x107
E (6872) diskio_sdmmc: Check status failed (0x107)
E (7873) sdmmc_req: sdmmc_host_wait_for_event returned 0x107**

eduard-firewave, 2025-01-02:
It stopped recognizing the card now. Tried several SD cards (all work on the PC and are FAT32 formated). Actually tried several T-SIM7080 boards as well. And the same SD card is working on T-SIM7670G board.

lewisxhe, 2025-01-03:
A7670 uses SPI driver, while SIM7080 uses SDMMC 1bit mode. It seems that the problem has been fixed in versions greater than 3.0. You can test it.
https://github.com/espressif/arduino-esp32/issues/7373

eduard-firewave, 2025-01-05:
I am using the platformio with vscode. I upated the env to
[env:t-camera-sim7080]
platform = espressif32 @ 6.9.0
framework = arduino
board = esp32s3box

and it solved the issue. 
Thanks so much for your support!

closed 2025-01-05 by eduard-firewave