Xinyuan-LilyGO/LilyGo-T-SIM7080G #95: GPS signal acquisition is taking a long time

FallingSnow, 2025-01-04:
I have my device connected via USB and the included GPS antenna attached. However I struggle to get a fix waiting 30 minutes and still not getting a fix. I'm using the `MinimalModemGPSExample`.

Where I have my device is particularly difficult to get GPS signal, since my Pixel 8 Pro also struggles just as much to get signal without downloading GPS assistance data. However if I get a fix outside, then move back indoors I maintain that fix.

My question is mostly about what I can do to help with signal acquisition. Should I get a better antenna? It there some combination of `AT+CGNSWARM` or `AT+CGNSHOT` I can use to help?

lewisxhe, 2025-01-06:
You can only replace a better active GPS antenna to improve the GPS signal sensitivity. AT+CGNSWARM is useless for this module. Unless you keep the GPS powered, turning off the GPS will completely cut off the power.

FallingSnow, 2025-01-09:
I have a battery connected. To keep the module powered do I have to avoid calling `AT+CPOWD=1` and `PMU.disableDC3()` when trying to deepsleep?

lewisxhe, 2025-01-09:
Yes.

closed 2025-01-10 by FallingSnow

FallingSnow, 2025-03-04:
Do you know if it's possible to just keep the memory powered so you don't need to do a cold start everytime instead of having to keep the whole modem powered at ~30mA?

I assume it should be possible since PSM claims to use 3.2uA.