Xinyuan-LilyGO/LilyGo-T-SIM7080G #96: Errors when sending large data over TCP connection

eduard-firewave, 2025-01-05:
I have a large (350KB) file on the SD card and i'm trying to send it to the server over TCP connection. I'm using the 1024 buffer to read from SD card and then write it to the client that was previously connected to the server. It writes about 10 buffers without an error and then starts to return the error response for CASEND command.

Code 
```
        static uint8_t buffer[1024];

        logger.infoln("Sending Audio Data...");

        while (file.available())
        {
            bytesRead = file.read(buffer, sizeof(buffer));
            bytesSent += client->write(buffer, bytesRead);
        }
```

Output:
```
I: Sending Audio Data...
AT+CASEND=0,1024

!���`P�ŷ���2c��3�␡ZןW����w�lCd5우PX�2�K␇�a�9␟S�(�]=�,����_��h�/0}�&���␁�2�t�␝ ␁␟␐����q␏&�5�o�˙9����␄�E��MaO��Y␏�/�+␄~�W�2�h�K␜W␞XJ��d�����}
�Ǣ`W�m��wa
>+3��␎�ѹ␂\�
��␏P̀����)�<o␒R��x,�␞␐l��␇j�҃�Y␟␋���$vd�ˊ���1���h␙ݗ␏␘H�␅��1��␖�␂�P��VA��ߧP��0�MzJp:���]p��H�s��覠�␟�Yp*␗
        �␛␌␙b!!��8*���`��oUk␔51E�[�8N�!T�U␂h␕��@�␏��␇9���|�;0f���␚P5�u��\G��ל�z�␛�D8␜�␙d`���$␒���<�pV�/u�#��j�^��[d�'2Dg�Jlw��  ␞G�␜��v��t�4␒    ��:�{�<␆�M��������␙     X␁��v␛�R0*�l^9␚4␙^����␓�␐M␞��yS␜��␆     e5����␞͉�rw��a��J␕�␐J��Y����␑����␔f^���␔@Ec␐␒␌M8␘�1swߓSA��h�"�|�������י8� ␙�Fl␕��,        ␆o���␁*z�␒␡y␋i␃!�ٲ���B,�'�hD�~␞���.Pl��"f���␖␒@u����
���U��&0�J/��)␃`��S␀0r���`�␌��ݪ\␒��(␑r�␚r0v�␡М��n�[�ӟ�;���1�䠣b��!��H�݌Q4����.�w�jF]���␔���␌8ekB��␒�|iV␕��␕�\�*␎�|��F���^ΒE�h}␇
{��)��␌�␡��F��␀␚�eGe
��]�(N␀�Bv��␛␝�"ّ␔ư�M׻␜␋�5]!������␟AE�␂��␅�␡6`��ӝ�x���%�␆�$� 
OK
AT+CASEND=0,1024

>Q@��+�GĞ��␚␘�l�NZ%��-�␞�␡␖�$bU�q]�T4,w�vm�␋�␎␋��$G�Gl6i���␇�␘�␌��ڂfe�1Ou����x␞)��9�;���␐��7�H)>�f���␂B�������j#d22l��PC%�>)@j
�Y�]��l�{�-�����]�s�\i␃Q_<S␄�s.�␑␗����ڋd�␑�"�␞g�F x۱2␜g͉␐�o␖��I��␅L+�Y��␎�x�:␕�␟��␁����BM␄␏��r␜zbm�];2�y�E+�48
$������␆�␋␏8e.�qd�#|':3��Z���/S-L�R�R� Egi1�HS��␀�␂␗r�S�-����c��*␔qB����<:x�خ���␖␡\����␃r␘␔
K␝e;5��␜␖2"���@*���|�U��ع��Z�_�fa&�k׌^هWB���ܼ�#�␇ܮ&6Z˜␘0␟�-�[�Xw��|h'�ՠ{␕s�␙��D�L�]JC�����+�?�T������␕␖s<Љ␔␇��4��},��5I��b6�J␗!���vo�@m�!��B�␟H�s␘�!\���#̩�␋�Ħ&MZ1Q���T�ߐn�k<2�k�␁��5i�V���`o����$n~v�␌��/�䲩&��كH␗�B$��␇�߸���C(�␏iu�L���␅_o����␛␏t�␎4�␖␑�J��,X�I�(��[��␋�M.�g�␑D␝��ҧx��   Ei��el�t~D���K���␡z@�G�␛�ז���؛�հ���g�E�{␚P�$␕��`Ȁ��r$�޺z�H�t,I�␄d�{s^ghf��␔���B�<␖��R�6Z��lG�:����q���␡�␕ac���D�s|"
�␑��hU��;��-�␂9␂k,F_>� 
OK
AT+CASEND=0,1024

�␟��␛WY}P�I�wO7*����^��␐��␎NT␏�1������V@��,��DU������F��2�␝R��␔^T(����_␙YM�Y�E␙������p�w٢�␃u�w�
�M3rr␑�H��,���␞�Z�␏i�Kk+�6}�<�!���n�␁�c�`Dx�0�/A␚�O��␅��Fq�M���&�6)␎␘�I���k��ރhQk*��䧍�`s��ɡ�j�H���{4��x␖�wY�8␖��;���>S��K1>]��Ќ�?4O�␀`Z;␀\�~q�␌by���P�u=����f��v�␀��_�qB�Ǌ"�ߐ␝9���g�ky␞����zx�ůG�(␄␆␡����f�DS�k=ԣ�␖4�}�<%i�␓�␓����)�.��v��␇
���?@q␀�M��M�␌F␁=��,�G��->��b�␛�q2��N␘s�␐�}
��␛�@������oz�"f��ir(␖I␌�e␐�)␝��|����!�I�Vm��F��x��M��^��Tr��"␂�F��␖␚�f.״
�Ƥs�rὃ�-t�>.�h:�S)␜`�|U�␑ञ��f�ѫ�A�s␆�!*��␇j�␟␁��E��*��(=�␌�Lu�����H��Q9_kw�␔hXe=␓ӵ␚4�M���␜yf��$␛�Z���A�n␛�ݍ�E)K���5ٓ_��d���      ��␄�␎~�␕␘␔␏�7QNf��       JDH␄�N�K�r�e��ݴ���a}�
ǻ␃�.␎2��b(�A�␃�s��␜Q˱␜ڍ8��&Y␃J␇ u�c␅␚��␃�f\˹�␁␡�␗�␓␚��␀�␌A��␜�M␛j^␖1;�␅�␀��8sQ␀��␞␡��␝���2����~B�t␙�␞␑␀a`?��֥G+�H�S���
OK
AT+CASEND=0,1024

>!␎�If�+,
���␎g�Q�        �P'�g��w�t�M$�<�p��D7 �V�-"j�G�H␓������US␘����1�T␇$�n��2␏�#��^��$]␛�␖: x%.�␟>� ����␖�f�␝,␜�O��r��GV3�.3␔^X�␎�hOy�␖r��g␋�
�(�bf�_���8fJ7�␗␛"ʤغ��6�6���␆��eW (a��␁�␁�␎�d�tZ�h�␋Y{xE6{ �\>
�uMK��+"�␖��␞��=SE␕��   u��^��␒���6��(4B
��Yn��{���rJ7j�␅Yp␂�߀�␂�␡/␌�8i&␇|(��B��␂�;M�}�������t�%��b�V��\␆y␐���/�␅\�[*���9�*␎θ��>el�ۭ��␇"Gn�����1�HO␡'Uvڅu��␌����/F;U�:�␞�?��=;���o>␚�������␜␟�=p�a�K3�a�(��9s� ��CP������q��'_y���]HĐ���d�␑����ل�l�R␝&`B�␑A-�␒*̈́�2(?.␝"=�L_�F�␔
x�?���e��k�␇����␌␎���~��/�:�qÊ�␐pӥ��txt'=}��I�␕dc␎�;␘AM�w�u��y␌6����␑w␇"�O��3yߍ␛�H␏,dZ��␖a%<OT[C7�b1�u?"sfo�Gx�8���␋\�þ�G�h'␞␔�␗޲��Ω�*�T��}␝A֯K��SE�ӌ␜��␒R��]��Lw�gֹ�';0��␁7`�␁�[�␚�␓3@�Y��P瓢ϻ����|␎_1�d�~��/k��a�+�␁     �Ö��␌��␏w�ӊf��K@␔���b�|{5��Qk�GpX|�␋t�}?a{Ȝ㡭␒��#<"␀܀�m�b�␜�e�c�a␛��AmF�yR)��    ␎1X]�{Ƃ*7,�<␔�RDʈ␆␚␆0��=��l␋␁␜�1␝pA;�␕E&�`6��␖�9�␁�ۍ␋Փ��|␅Z��R~M        S�4�f�R␟␃ 
OK
AT+CASEND=0,1024

>␗��␁�M␙��B]�6␚␁g�$&@M¶␘˜�7��#2␖��!�05W�␘C␟�Ic��>8[w$␜␡g�␑��j�␘�A�␘O����@!�m��I�␙h���V�d��␖�+���L����Jϛ�x␌�}ĵTtW�\�z���K.�f�␞z�Ѓp�4X��gB␞�␀A��ȇ����␘␔^�k�e�����&9␒␅␀7���(���ꮍ�7�K���␒��+f�/LN    �,����K�␛�J�␟�g�D�|"����J~␟C�␒���9����{7�T�)�D����}��O9s␘3o��۽��s�!�␑c�␕gp�T��R␘k/�N�mR� ␅�␑�t�H�Ol��    ��=0{s��:F␡��!��/#�}�}#��␅#␌rB␗~���$f�h��*␁�,U␇�|␌��%�'��6[O0�v�er.}␙�l�_��p�����%��|~ԕHaU����␖F�k␃�p␛��*␗�qJ��!�Q�����B�4`v���[ll\V�m␑0�[0.*������␎��/}C��y␓�|)84���␙0|�␚77ܮ�޲F���1�#g␁|␔��␅e+(��p��%�$B�u�␡%��x
�f␜␒;�^#C␂␟.�Fs�)�\�␎␇!�ස�5␜�C���␐���␒L��.V�׆�R␞M.@�\t�ok~���T���g��B�wL_N/'���␄��&GX�␁
�ߜce�6��܂��␌x�9���Y:.w�ͯ␙�i����␁[␔�!␎zk��k�L␕␜ӳ␟�\␐␛�␆h@�␞�<4#��␒�^␅.�YUt;f�/�r��␐\S��8���
��␀���␏�i�+␇ʌ���َ�԰��������
��X\����\␕
�U�/SQ 
OK
AT+CASEND=0,1024

]��TBj��8␜cu����ŖwN␟t␏nkK��␚1���V�␡N������3�w�E�7©a␟␎fl␟��␏d"␃�k␀r�␐�v��qM�ݾ��I"␃��␞␎␙�g�dM�)��r�&��m�e��J␖␑���BbT�␁�4@�L��I�/  �%␂#�|O^Ѓ�␑␋
k␏=�W}�XX�נA���dK$�!��␟7�(��p␙���lU$dO�,��uf␙<�␝��􊗧��␝�+����*C��Cj␃+ɍ6�U����␅��+�M��␂(�6␓��Pn1��␕��N␂��
�;���HcM<��ƲH␕�,��nC�u��_Q�␖␋�{ᆰ�␙��]]��}�f��L���U␑�]␛␚�2���\Jf.�pI�\�-� �T�4␡=��;�=�kJ69<��␀~K␌0␋Ǌ␋�J�i�����[␆&���[ْŖ␓␋�ٰg�:m�␖O�����␜��O���H!���bp␗�"_�Mw��ad�ݼ6g&����␇��q*       ␇M␀����_�w/�␐��9␝Ko�,B�Uy��Do;��-x�:@/�␒aH��6�yc�.␓��␒␘5&1gN=���␛5D*␔ע�M�A5�␋␑�2��<�'0Wg��d8*�
�$��T��3U9��j���w␛^������;��c␙rH7`��US�␄1a�T<�v�ą�!���29�"���␏i���␅ULj�`n������(��␕�����>�֥�     �E�$:���M�X��\␘��k�␜u�␡�{y䦡���␎!��␖I8Ǫ�m�␁/ �\���␜�␃�əe���@␎5g3��/�8�/�P����;e/��#D U'␂}�␅U�_�ݘ��r���`�����}Fuf�?=)���␁b;��4/����e�'��$���}�VR�␞�����␔�U��,n␗7$y�␏P��7i� v=�V�)v
[... 72 lines cut ...]
/��8�,%!2�␟ԅg�L�␂��␗l2BK␋#␋�V���\"�ҹGw�}D/e�tz4�E␅.&�BL$���v�r␡��v��I�;��fV�֑���␟'EI�йr}␟�Y Bs(␛7�v���m␕ �D�-Ʉ
ERROR
AT+CASEND=0,1024

>��fE�"�<�qMb�␕s8�̻`����\ac{�Fr�8��␔��X␡���FF�w''␏+�㾺�`␔�S�3�޼�␛�6����X�)�׌r�ό␁軰*␓ �IK.-0Yۮw0��Hol␌Y����!�����(�=�n�␚p?�s␟;-����H��mU��n�{�;;/�d<��#n�`␐␌��j�␒��_��␚I��␏␝M����i�FrŚ␛�M�␑�␀3�(7�#��1h�>␝␄"���,e?L���[���␄1���^z␖␗]Ic�␐����p�␂$d�␐␇��n��!S)C�␗LBz␃�"�<<X.�̰�␘��_ՃN�␋=�FFD�6�␌�?␟g��eȘ��k��/+0�␓3;�␀��z�VG�Y�Q iW!�Js�^�B7����␞�̏��X�<�␔��䯡N�[���N␋�2��wvL���Ex���^@OIm��mY.���'�"�8␔�UV
J��e*�B␏u6�VFHz�ec���U���␡␂JƄ^␗c{U�Q�Av7␟��oe�␕␎␜$.����,��␕+␋␑�{~��␎�2�=␕�g�d�#��m�␅����ճ_␓��wrMR�      ␁#J�W1"���D�{�*��XbMFf␑]��#��#␃�[�␁�N��␗��␑��␁���␝␞:AQC�ܧ�␀Uٳ]␋f��$�̲�����i�� )␜C�0|q2'␖�7�o␎^��G^␔r�O��|2i��v�ļ�oO␜��d�␚D�!␏�.␗&�0�␎8��␜�␕4U�`��[wiq�\��["�� 
ERROR
AT+CASEND=0,1024

␆t�␘�␃�]^>
�a�;�␚7C4�rd-����␑p␔B�d�|�?␝�6␄�|<:�␙␂$w�(��r␅�␎�j!V���3ې�������y��0��:^�L␒␜.����␖�In2��B���a��␗�tH␙��␙�k��\␋:������␓␑��/!��␐b� K�k\�Haޡ��E�e?
Yb4:�Ӷ9���ȆW����C!�␗��{�T␒s�?<-␄eƍ�*S␁��2".З�\␒�_W␝a�����␖␜�Ų��␐�v��扦`W��zv�+7�ω␀;谤�␒����␃�%'m␇���A�␑�i����`���␏�q�*��&��x�␜��j ��␡`��>Z�mf��"�m��B�>P)�C�␝�(�I��ݙH␟q%�"K␖{�:{2�^�␎"�F�␅�JԦ;0�c~w[��T␕N␜,␡)� �IR␘�C�␃␐��'�T�␋�(
�␛CM<(�@)<�+�␋щR'␘␐Z(Ͽ���S}wī[���␀q��8���␇�$$␜���r5T���O␎q�h>���␟)a�m'9��␋\��K��␐V�u␄�b␇��ȴ.�␃��\�␌kpp�]��␓␕TB�E:~�5␙^,��z�����]��'�ƒ␓ɳ��R'z␐�1␁�E�ӖQ��$␡T�␌�r  ␃M���␃1�i��
�j�I␕;��iL␌&J������r�vB��/�f�O��Qz�~u`�␑␛�␑���␗!�y�h(g���@=␞S'�␄X␘U�(K�X��Ƹ��M��`ǿ���D��C��˽x�[UKs@ܔ�␖�*�L����7%�R|��^w��J␚�*E�␅y%��r9%␞LZ�IGy#␘�FF��c�8␀�h␂��␓����1�Ʌ␐G�����␟�4=2络�␝5␁7�D(�␕��␡ͳ�k�Q+�k�E�[G����ԫ���~ ~�␔׺��-]���-��E�c|I␂␡tЮR�␓␒T�x)�␄nj�s�O�w␗hC�:��␟�B␓�P��A��_�?�o␓�ɞ�*��b␝␡�Nc԰��W_>���##kȖ␋��1 
OK
AT+CASEND=0,1024

>V�乫@g{�5���)�␘���␑n�␀A��␛ڇ    ��␏�+<b%G�␋�ibOƵ��_�{1�,�+q)&�r�2����␡�i�j.�#3���������+d�J␡���N␂�␑;[���3\z�0ΟsNl��:M�␡��m��D�he��cm��qO�J�.␜N␎���␡��I␟�!e�?�P�m$:i��3�������0m}�-
QE�C␌��␓␜�N��␓���S�␛�␚␏�-␚␀�m%�S5�␗     '��q�q��.�����f~��␋�$=m%�d�eq�x�3␏o�=�#��␀R7������y��$l��␌g␋�B�m␔ ␐박 f"��␚␜��s�r*h␚�B���2␒�����␘t�sp3�vc
k{��␂m�␃�ף␙��p��,Ì�YA�=Yf�%O�&�/�!`�␆␇␐*�␗`i�␃��NV�V␐;`C���␎�"`��e␟\,���4.Q␘��V��猫�k␓��␑�7��␚���:���=�>m�␗��$TdboVW2�_j~'�t    ␚␛�W �8Utty�̯␜#��<�␄-�␂�3T�w8MO�0��f␔�6D�E�/�\�␘w۴␆�␙�4�-o�/���-i␅��␜[e��␅␙
-�␐����g�$�␀��␂␚��gP(b�w'��␙�mJZ�Z�␁�z�� c䠷�]�c��ܜ�L��|��p���Ȉ"o␌�U�� �?␜g��S�x��3␜��?␗:�3�����;�k�A�H���W÷�j*W�*6@[␏␑��
c!�|Xb�8�␡�ڊ␘~@�J��!Z��l�␚?�Y?�n�zo��2�q�␙~��␌��Ra=��^C␕SE��룢���␝���t�4��FG*ԁ1�pT,��U�%��|l������늳��(�w�+1�␀���r�Բc�����~1␌�IuDwˑ��7?�␝�R␁9␞����~{��9���u�R#�|T���(�RB9�o�XR����x��h ׿^�␕��j␅j�f���漬n␀s��=�1
"�␚�8ր*Ě�>=�(��ɘ��F�␖��c��U��␔�<r␁� �#��n�l� �$�#���B"�Iq␏�� 
ERROR
```

lewisxhe, 2025-01-06:
There is too little information to determine the problem. Have you tried another server? Is it possible that the connection was closed by the server?

Send AT+SIMCOMATI to query the current software version and the PN number on the modem sticker. I will check if there is a new firmware to try.

eduard-firewave, 2025-01-06:
The server should be fine. I have similar code working with this server on T-SIM7670G and on another esp32 device working on wifi.

The response to AT+SIMCOMATI 

Revision:1951B08SIM7080
CSUB:B08V01
APRev:1951B08SIM7080,B08V01
QCN:SIM7080G_P1.03_20210823
IMEI:860016048838077

OK

lewisxhe, 2025-01-07:
Try updating the firmware and try again
[1951B17SIM7080.rar](https://drive.google.com/file/d/1-m0eF53mw9n0vw4vfNZ88cRR3GM0T-a1/view?usp=sharing)

eduard-firewave, 2025-01-07:
Thanks! Will try and update.

Can I ask where you take the firmware from? The one I got is from :[ update_simxxxx_firmware/1951B08SIM7080](https://github.com/Xinyuan-LilyGO/LilyGo-T-PCIE/tree/master/update_simxxxx_firmware/1951B08SIM7080)

Isn't it the official place to take the latest firmware from?

eduard-firewave, 2025-01-07:
Unfortunately, firmware update didn't help. I noticed that it is always first 9 packets of 1024 that are passed successfully before the errors begin.
Here is the full output of the code. Hope it will help to clarify the situation:

[  5571][I][esp32-hal-i2c.c:75] i2cInit(): Initialising I2C Master: sda=15 scl=7 freq=100000
Mounting SD Card
SD_MMC Card Type: SDHC
SD_MMC Card Size: 30436MB
Initializing logging...
I: Logger initialized
AT

RDY

+CFUN: 1

+CPIN: READY

SMS Ready
AT
AT
OK
ATE0
AT
OK
AT
ATE0
OK
ATE0

OK
AT

OK
AT+CMEE=0

OK
AT+CLTS=1

OK
AT+CBATCHK=1

OK
AT+CPIN?

OK

+CPIN: READY

OK
AT

OK

Modem started!AT+SIMCOMATI

Revision:1951B17SIM7080
CSUB:B17V01
APRev:1951B17SIM7080,B17V01
QCN:SIM7080G_P1.03_20210823
IMEI:860016048838077

OK
I: Start modem...
I: Testing AT commands...
AT

OK
I: AT Test OK
AT+CPIN?

[... 240 lines cut ...]
�c��s�ع␌Ƣ\߄�0,<�`'b��{Y!�e"cF�␞�ҡ␚�"^t� ֓8␀2�s糮␁␁ 
OK
AT+CASEND=0,1024

>␛�^)���[��sX�-�����P␜fK�ui�␚S���:␇AH�ѳ��s�q�1�nP( W4')N)�3/|W␅�␞|␜ш�t���␜�'␗�␃bh�R��6���␌�!s␔yX�*ZQ␕␎����␏H�=�␅C�ռN�-ڂ ␔��aT�␌����␆#7S_oR��c��=1��␙=O��0ٶ�␕␛|M��
a4C�$�O�7n�8�Mw␘␎�`��w����0␖␆␔␌�j�4�����5����*E��d3�rQ␁�:�␅\␔b␔������1ώh�y��VN␜␂i��b+r�␟����P.␂*P]���␒�8D�t�v��␋oMp�h�o�␇rS���?␛␄�#��␗��e�s$��Qbb��?WT�␛v��U��O5�␛y␃�Q��␒%|��3�u�/�␕�֧�� ��ƞ��0��␡␇p�^(
�B␚a�m�:WW��t␡�␓Uy�L��_�n`��␃��r���O���$�^�)%�`␜�Ԕ���␏␐Ȃ��2�K���␐/��f�␇G6␇␘�␙-����␆�:��␖�␡␞Al�I}�2�␇wlhx��Ja�␗x␀␋�O␎␡ׇ�`��␂�␅    ��␖��%�a␕�␆2��/��=��ʡ�;D�t��X�␁�␋�;␞�ʒ�s�W-Q�e/`"���8�.U.q�␛�}␟�␋^L%:V�␡u[H␎�(��Ԙ��$��␏�ӗ3j�k�ˋ�␙��␝n�BU&j�H�һ�[؁�)��"␛␑�O�Çw��T␀�L␃y>�D��
JI5��L�␄�7K�V�`d␏�␅�␚�К␕bwԽR�&C�X~q�룦#�␑)���Y��sc���U�>�␚�=>[␁�$�␎/�?,␡ߤ�q␏2␙��Y��␛␒8o��@�?␕C����*�␎   D��xS4�␐i��C��D��%���0!��#<*.T=J��M)�Pʯ=��co%V   Z��Q�t�7�bAt��,ټ�*&�␝�␁(anX�u���c̮4h␀␀��UΠ�r␑␐�k�D␞�89y␘�B       ��␎G�,����␏Z(␛␝H��3t՗␞V��c␌�Q3 
ERROR
AT+CASEND=0,1024

5��$�␕��␌␅�g�?��#�ęuP␏3�C�'`�v�<!�-h�v��␜P���T�␖�␂␂f�jZ��Q��,�␁�␀��T���َʟϯ��L�Sk�␜���-V"␎�>̇�J1��q:�_␐␄N␕z2}d`i��␅␅�9@�~qRr�Hgi�XW��",wA956�|`�z�AO�b'��␄f␇�����>��␖?����[�(�␡��
&�␚␅��U�D��␐z��%V�T!�ڲ_�vgC��"�|rz@����;j_B2Y�4�1�      �@��w�z+J㣝d�.�炯<�6^␛(�rm�(x֮A���ʃ�ӡbLvV�␒f��2��p�=pjj�dE�=␜gK␑�׏␔7j��u���wO␆␎␡2�=␌?R���␟cN��Е}␀��U���J<v��<��K7W7&�␀��t�␡␡+%Ų�:␋��␔�$��A>␞�␚␜�������Y��m�z␏P␌Z����␔���ߨ�␃␛�<�␐} ␓B���]���␅�z�ɞ���3���أ�ć^���5�ܶ�%␃␞␜␡��3o���`,�����"Шu�JՃ`B��!Y_C���+F␄1/�d��ӆڐ�:!|␂␚�2ɞ_:␆���"�R�o�/␎��`��{ѡ��D���␗␂�����CY'�L�۸:���:~1B@�}��㉺�o�j�2Ѿ����␕���^f��s����o␕𑦘#��Q7␎Q�␋�␋␄cB�~u␓v     �␙��    S␅��|Wk␅)␌�@���)��il}��␒�D���el��E�J�{�sY~����␖��␄␘�)�ӝ^?␐J塐Z�␞␑␋�?2Z�/ְ�_�����␇��*H�_�␗␇�T␔A�f��␞�~␄�_�������a%� 
OK
AT+CASEND=0,1024

�I�␆�B�#,ʱ�AE��␌Aš6㎶��<Ő��H窑/N␅���Lktj��7H�c�␗�uO�,N�;␘��7j��L␋␆␄�Z}Ê�Y�kr'���{���␃␟|\g�'�����e��2�␏�
l����Y��WNt%^0�Lx�]��_>h␙�#t���␇)&�^0��]}␙���Y.-2�$␛�S4���=e␂�k�ҭ�␔�N�É+&n�␝��3␝��3^��*w�˰�     �-��bK�HRx␌�B`��␝h�,���w��p��s����␗8[.}���_#�␇�␄���P���2'�x␖:��'��)*���e␕�5SU`␇�U�v��␎چ�␐�"�}��0�[�0␘�␑o�l���y��␒��Ei���pד␗u�h�␑�o��␝�hFF�:k*噽`5␓␗"␌␎��c�1�T�␝c��FYp��q��␞u(␕sdx�{�␡␗Zz␅��G�m��Ș��^�␒X�'�>��␐��o�␗FS��IP�)␇N>(CWԕb(�␝u������;Q␘�GK@#�␁Ɯd��&��>␖��!�uU�cOX9>$5��}aA�Av��Y�␛␗�v␋-2�␀���1�D��V�5 �␡<␃��l)��ҹe��\��j=��␓��6���s�E��v␆�+�ɗ�␕C�P6ڏJ�b�␌d���t�T␚␚�G�C␓�hx��[G!Õ�␋��h'$␁a␇�qA�␄`�p�o�J␛��Q�b�-k���ʩ�;␚nd�k��.f�A�<Yx[     �A�y��V�␘>K�␏^��␆)|␟␅^␖��C␄Um����^��␙��ӑZ6�S�g��␝�␆3�4��ލ��+␜]>V��#␎␀�ҪU2�T�����Z?␐␞H4-�R�"␡␙�ƀN\�:�DC���
�G���r�k␕�␙�Oφ�p��hV������%����*��n���␜␁��␐R�U�RP`a{�6)p�␞�]�|.��؝␘����%�.��؃␐F"�Ħ�}␅&4�7�'���, 
ERROR
I: Total time to send file: 2040 ms
I: File sent successfully. 12288 bytes sent, 0 bytes tried
AT+CACLOSE=0

OK

lewisxhe, 2025-01-08:
It seems to be normal. You need to provide me with a capture modem log, and I will send it to SIMCOM for support.

USB driver can be viewed [here](https://github.com/Xinyuan-LilyGO/LilyGo-T-SIM7080G/blob/master/docs/sim7080_update_firmware.md)

The video of capturing log operation is here. [SIM7080GCaptureLog](https://www.youtube.com/watch?v=AEIriATGVb4&t=4s)

## Capture log
[DiagGrabPro_V1.9.7z](https://drive.google.com/file/d/1-lnNPaYeiPiAmtlAELLkv4R3_I4lSNfO/view?usp=sharing)

eduard-firewave, 2025-01-13:
Thanks! Will check and update

eduard-firewave, 2025-01-13:
Here is the capture:
[20250113073144160.zip](https://github.com/user-attachments/files/18393714/20250113073144160.zip)

lewisxhe, 2025-01-13:
Did you run the same test program when capturing the log? Can it reflect the error after more than 10?

eduard-firewave, 2025-01-13:
Yes. I ran the same program. I sent a smaller file, so it had much less errors than usual, but there were some errors:
I: Connecting to server...
AT+CACLOSE=0

ERROR
AT+CACID=0

OK
AT+CASSLCFG=0,SSL,0

OK
AT+CAOPEN=0,0,"TCP","data-collector.dev.firewave.earth",6000

+CAOPEN: 0,0

OK
I: Connected to server.
I: Sending Chip ID: I: 0x84a43543ca48
AT+CASEND=?

+CASEND: (0-12),(1-1460),(100-10000)

OK
AT+CAACK=0

+CAACK: 0,0

OK
AT+CASEND=0,8

>␀␀��5C�H
OK
I: Time to write chipIDBytes: 28 ms
I: Sending Audio Data...
AT+CASEND=?

+CASEND: (0-12),(1-1460),(100-10000)

OK
AT+CAACK=0

+CAACK: 8,8

OK
AT+CASEND=0,1460

���(mkIR��-���7�'␀�$�Nΰ�M�5����}�T���␆␆␒x���-zp�Y ϧ$k�� I���<j␀␏S�␅������p�݄��␀�US�QA␚␟␖0␔�*␛␛K␘����.iT�␟ܧ�u�//;�␞9Z���␆���H��n��nt�␋␋[�␖��ՉE6u�C���%2�����␇(�R�#CF^'�G␃ˊS�9␒�⿾W␓�     �9�MQN�{kY␃��bj,�%"��␎�~�8Y-�nE�R�pb␑C�.e������c5���t8�L��Z:�9���␃�Z4��/i�6���n�g����ـ��y��k�����␄_�'K␀0␋�G[s֓�(␝y��t��\
Gl<��ե�ۣ�HQ2imz�␂(]�␞␡,�2�o3���D�Py��␁���v��nd�d�␋␕q��␡␙Y@6*,����2�{o?�S��␞���5��|�I�␕�޹�l1�[2��1<{,̌����x."�k� �Vִ�␑␋␂٣6+���j   F��䳘5�T^}�0j   0V␏��^g�!␐#,�+gl�����
��#퍕;␞�S␞�5�H��%�j��恲�␙T֑␏␟��␖O␁�1>$�r���␞��%�}�P�EBa�C����
�␏�␑8�U�F1��␡#�7�j��␘�␛��פ␏g␃:Q��uydo␂�qzR�قFT�a␞�J��N␓/␑���xP���␁␃�_�1␐�>J��Jo�␘� ��#�.N�␇␀x͟�a�␃�)D��F7␇��#5␇␇�h݌ǥ[�/��␘␔���HSJ␃+␌�#␌�%u���␓8_eM␚␒l"��␙␡n.���ЈDZq"B��␋9迓m��9�g#�4L��س���Vj�␁��㭇�l��<R�;␋r␐���r�MP6�␞�s�[�b%�!w ␁␆␀?l/����$t��f��J�|O(�Dņs␌�O�}�^␀�s���␌�␌
{P�;��\ɎW�␁#��␇c�)#yr��Ī���F�my����A␗␚␒␇␁5�,ԓh␑t␗Y79zw␛:.       ␜␛�m�A5'Q�q␄s�^]��޻�2�␆)�␙l�S�+o��D�³ּ�~;��BO␛��␇�57֣�␖�␄��;T��{rذ���N�P(�␞���,���'a�|���GR�p�2�Z5@_Bp␐�c��␏��␏�9��F4
3YƉ['�␃ej��L8�␁$␃����␝�f.A�'�E��xX␒|����␇)
OK
AT+CASEND=?

+CASEND: (0-12),(1-1460),(100-10000)

OK
AT+CAACK=0

+CAACK: 1468,1468

OK
AT+CASEND=0,1460

�i��␒�a␇g␜>o�)wč5h���-␞����␀���o���4)��
ɯ����x9�����Un�Q��;�"1�Ï�͒��D^���␙�k�ѹ�4�␏h��Oj␙�br�␙�K�O`
␙G�8mCz␀J�␁5uy��dX�{��␕N���Dw�  sߎ�7Q�?�␅��|�d7b1n�$���␚␇�␅��Q␂8���Wb"!򚩝 P�ն?>␇��C�lS���2␇��u��+�)e&'�~D␑�����␇\T��?�|�U�o���␃(␟ňA`5�7 ��Q��␛'��}�,��졘␡�␖��␕����)<P=�A�R��p�,�3!h�␚h�xsgA]_v��F␆␚$��a��␙[h���\Z␄��9H������␓�j�␁qK4�␌A�#䪱ml␎L5␎y�}Q�_�-n9F�,�␃��">˱,0e␑��␋56Cg�����#␅<����S�␡'�iی␛~�Q��)x!␌]Uq���␐)���9���␖˗�لc�2?tV�T�+��qI�P o����p+�␅�Q����,��AZ��[�␗�zV�␇�@���~�e��J�␟␆;����]�6'��b�o␙��:��!Jf��␂:iԪ�␑Z␞I߬�/�/���␀���2�����␘��qUv"o␃␐�oo�␟␁�����XK�m8��.�␅p��@�R0�P��fэh#2�0�e␞'�w�����F��骁␌^␑Ar]mW
�␃;:G\���2փ␓b�+␑ȯ��␂˯����F����  a��␒wF�j�']m6�␅�~���g�����␔�.␁L␗�O;�^*���␖�~�␋.%r�]����␃ˁzpgu�␕e�@�Q�Y%e��{&q�s"�Q�<��[J␛�8��0��]�␋��.�G$k�␘��kAW��}�␒␜6:[f3�S`T�m␂��p��f"���5qd֝�8����Gi-JE�je�p��N�bY␒H�)E��7�%��#��
�̷`�S���QE�␏�y4�6���.,>`]ȼ␗GN␄�␟�O�1�␁|<]�~�B�]3�*��␚�0is����}␁&"h␓␘␆␂$`Æz]JheI#}��␘c␓�6�����}a�cm�6ɳx�����'K����4�Q␟g���/�␜����&�,�Q8$�h;�19␡tc�c␙M~Q�������G␁����A�vd �Ur^8␆�1�y␇�%����␏�����␟␏�4W\�]��␝�a�A'WU␁6�y��%��T��'_M␔�␋:`Vx�!Z�o=V�%���o��␌␐6vQ鲛x␗�i]��Yj␘j}�t�␝)ۡ��i�I�㮕�$X�0U>$�␓�QWv␏"Ed;��Q��Q$Rݗ��␞�8@�Lр&�l�s~烥�f�␅��%����a�Y��<��ִ�9EǑ�P�~:��_ 4�L(g���=���S�`��8��1␡�d��␀�
[... 226 lines cut ...]

OK
AT+CASEND=0,1460

=␌?R���␟cN��Е}␀��U���J<v��<��K7W7&�␀��t�␡␡+%Ų�:␋��␔�$��A>␞�␚␜�������Y��m�z␏P␌Z����␔���ߨ�␃␛�<�␐} ␓B���]���␅�z�ɞ���3���أ�ć^���5�ܶ�%␃␞␜␡��3o���`,�����"Шu�JՃ`B��!Y_C���+F␄1/�d��ӆڐ�:!|␂␚�2ɞ_:␆���"�R�o�/␎��`��{ѡ��D���␗␂�����CY'��L�۸:���:~1B@�}��㉺�o�j�2Ѿ����␕���^f��s����o␕𑦘 #��Q7␎Q�␋�␋␄cB�~u␓v       �␙��    S␅��|Wk␅)␌�@���)��il}��␒�D���el��E�J�{�sY~����␖��␄␘�)�ӝ^?␐J塐Z�␞␑␋�?2Z�/ְ�_�����␇��*H�_�␗␇�T␔A�f��␞�~␄�_�������a%�3H9�u␝       ␚�pc��␀�ȗH�$��11�I�␆�B�#,ʱ�AE��␌Aš6㎶��<Ő��H�,N�;␘��7j��L␋␆␄�Z}Ê�Y�kr'���{���␃␟|\g�'�����e��2�␏�
l����Y��WNt%^0�Lx�]��_>h␙�#t���␇)&�^0��]}␙���Y.-2�$␛�S4���=e␂�k�ҭ�␔�N�É+&n�␝��3␝��3^��*w�˰�     �-��bK�HRx␌�B`��␝h�,���w��p��s����␗8[.}���_#�␇�␄���P���2'�x␖:��'��)*���e␕�5SU`␇�U�v��␎چ�␐�"�}��0�[�0␘�␑o�l���y��␒��Ei���pד␗u�h�␑�o��␝�hFF�:k*噽`5␓␗"␌␎��c�1�T�␝c��FYp��q��␞u(␕sdx�␌�ף��␃VK���2␖���M~{␛�$�)�Y␂�K␕���B��y߷���V�yA�[0�=��퀓�&�)�{�␡␗Zz␅��G�m��Ș��^�␒X�'�>��␐��o�␗FS��IP�)␇N>(CWԕb(�␝u������;Q␘�GK@#�␁Ɯd��&��>␖��!�uU�cOX9>$5��}aA�Av��Y�␛␗�v␋-2�␀���1�D��V�5 �␡<␃��l)��ҹe��\��j=��␓��6���s�E��v␆�+�ɗ�␕C�P6ڏJ�b�␌d���t�T␚␚�G�C␓�hx��[G!Õ�␋��h'$␁a␇�qA�␄`�p�o�J␛��Q�b�-k���ʩ�;␚nd�k��.f�A�<Yx[  �A�y��V�␘>K�␏^��␆��iD^␖��C␄Um����^��␙��ӑZ6�S�g��␝�␆3�4��ލ��+␜]>V��#␎␀�ҪU2�T����)|␟␅␞H4-�R�"␡␙�ƀN\�:�DC���
�G���r�k␕�␙�Oφ�p��hV������%����*��n���␜␁��␐R�U�RP`a{�6)p�␞� 
OK
AT+CASEND=?

+CASEND: (0-12),(1-1460),(100-10000)

OK
AT+CAACK=0

+CAACK: 17528,8860

OK
AT+CASEND=0,40

>]�|.��؝␘����%�.��؃␐F"�Ħ�}␅&4�7�'���,
OK
I: Total time to send file: 2333 ms
I: File sent successfully. 17560 bytes sent, 0 bytes tried
AT+CACLOSE=0

lewisxhe, 2025-01-13:
[SIM7070_SIM7080_SIM7090 Series_TCPUDP(S)_Application Note_V1.03.pdf](https://github.com/user-attachments/files/18394017/SIM7070_SIM7080_SIM7090.Series_TCPUDP.S._Application.Note_V1.03.pdf)

Hi,

SIMCOM replied that SIM7080G is narrowband LTE, and data packets may be congested. You need to check whether the sent packet has been completely sent before sending the next packet, otherwise the data packet will be congested and ERROR will be returned directly. After sending a data packet, use AT+CAACK to check whether the sent data is the same size as the sent data. If it is completed, continue with the next data packet.

eduard-firewave, 2025-01-21:
Thanks! I switched to CAT-M and it works much better.

closed 2025-01-21 by eduard-firewave