Xinyuan-LilyGO/LilyGo-T-SIM7080G #98: Unexpected NORMAL POWER DOWN

eduard-firewave, 2025-01-21:
I get 'NORMAL POWER DOWN' printed every second time I run the same code. The module is connected via Type-C and in setup I have the PMU code as in the example:
```
  if (!PMU.begin(Wire, AXP2101_SLAVE_ADDRESS, I2C_SDA, I2C_SCL))
  {
    Serial.println("Failed to initialize power.....");
    while (1)
    {
      delay(5000);
    }
  }

  // If it is a power cycle, turn off the modem power. Then restart it
  if (esp_sleep_get_wakeup_cause() == ESP_SLEEP_WAKEUP_UNDEFINED)
  {
    PMU.disableDC3();
    // Wait a minute
    delay(200);
  }

  // Set the working voltage of the modem, please do not modify the parameters
  PMU.setDC3Voltage(3000); // SIM7080 Modem main power channel 2700~ 3400V
  PMU.enableDC3();

  // Modem GPS Power channel
  PMU.setBLDO2Voltage(3300);
  PMU.enableBLDO2(); // The antenna power must be turned on to use the GPS function

  // Set the working voltage of the sdcard, please do not modify the parameters
  PMU.setALDO3Voltage(3300); // SD Card VDD 3300
  PMU.enableALDO3();

  // TS Pin detection must be disable, otherwise it cannot be charged
  PMU.disableTSPinMeasure();
```

Then it tries to connect to the network and goes to sleep for 10 seconds and then runs again. 
Here is the output of 2 consecutive runs where 1st run is successful and the second run fails:
```
Reconnecting to /dev/cu.usbmodem2101 .....       Connected!

Mounting SD Card
SD_MMC Card Type: SDHC
SD_MMC Card Size: 29819MB
Initializing logging...
I: Logger initialized
I: Begin SerialAT at 115200
I: Testing AT commands...
AT
AT
AT
AT

RDY
I: .
AT
AT
OK
I: AT Test OK
I: Modem started!
I: Start modem...
I: Testing AT commands...
AT
AT
OK
I: AT Test OK
AT+CPIN?
AT+CPIN?
+CPIN: READY

OK
I: SIM card online
AT+CNMP=2

+CFUN: 1

+CPIN: READY

SMS Ready
AT
AT+CNMP=2
OK
ATE0
AT
OK
AT
ATE0
OK
ATE0

OK
AT

OK
AT+CMEE=0

OK
AT+CLTS=1

OK
AT+CBATCHK=1

OK
AT+CPIN?

OK

+CPIN: READY

OK
AT+CMNB=1

OK
I: Setting network APN. APN: net.hotm
AT+CGDCONT=1,"IP","net.hotm","0.0.0.0",0,0

OK
I: APN set successfully. APN: net.hotm
I: Setting network MCC: 425 MNC: 07
AT+COPS=1,2,"42507"

DST: 0

*PSUTTZ: 25/01/21,14:41:52","+08",0

+CTZV: 8

OK
I: Operator registered successfully: 42507
I: Wait for the modem to register with the network.
AT+CEREG?

+CEREG: 0,1

OK
I: Online registration successful
I: Registration Status: 1
I: Connecting to network...
AT+CNACT=0,0

ERROR
AT+CGDCONT=1,"IP","net.hotm"

OK
AT+CGATT=1

OK
AT+CGNAPN

+CGNAPN: 1,"net.hotm"

OK
AT+CNCFG=0,1,"net.hotm"

OK
AT+CNACT=0,1

OK

+APP PDP: 0,ACTIVE
AT+CGATT?

+CGATT: 1

OK
AT+CNACT?

+CNACT: 0,1,"10.57.208.200"
+CNACT: 1,0,"0.0.0.0"
+CNACT: 2,0,"0.0.0.0"
+CNACT: 3,0,"0.0.0.0"

OK
AT+CNACT=0,0

OK
AT+CGATT=0

+APP PDP: 0,DEACTIVE

OK
I: TCP connection closed and GPRS disconnected.
I: Going to sleep for 10 seconds
Disconnected (read failed: [Errno 6] Device not configured)
Reconnecting to /dev/cu.usbmodem2101 .....       Connected!

Mounting SD Card
SD_MMC Card Type: SDHC
SD_MMC Card Size: 29819MB
Initializing logging...
I: Logger initialized
I: Begin SerialAT at 115200
I: Testing AT commands...
AT

NORMAL POWER DOWN

OK
I: AT Test OK
I: Modem started!
I: Start modem...
I: Testing AT commands...
AT

OK
I: AT Test OK
AT+CPIN?

+CPIN: READY

OK
I: SIM card online
AT+CNMP=2
AT+CMNB=1
I: Setting network APN. APN: net.hotm
AT+CGDCONT=1,"IP","net.hotm","0.0.0.0",0,0
I: Failed to set APN
AT+CGATT?
I: GPRS is not connected.
I: Going to sleep for 10 seconds
Disconnected (read failed: [Errno 6] Device not configured)
```

**Note that in the 2nd run it prints 'NORMAL POWER DOWN' and then everything goes wrong. Can't understand the reason. **

lewisxhe, 2025-01-22:
Didn't you see the description here?
[image]

lewisxhe, 2025-01-22:
I uploaded a simple test case, which does not have the problem you mentioned.

closed 2025-03-07 by github-actions[bot]