Xinyuan-LilyGO/LilyGo-T-SIM7080G #116: About GPS receiver model and PPS Signal

RenzoFurlani, 2025-02-21:
Hello,

How are you? Fine?

Yes, there are many other boards from your excelent LilyGo products, but , now, we are looking for some technical specific characteristics.

These are:

We need a PPS signal from GPS receiver.
It is important to use a uBlox receiver for its quality, but we are open to other receivers, with similar qualities.
We need to know more about the functionalities of the Battery Management System, for battery charging and battery loading, and for know how we can charge this lithium battery.
We need yes or yes a SD card slot.
Lora is not important for us, we would use BT for near communication.
We need to comunicate with external ADC devices, using SPI protocol.
Could you met me with some engineer from your company, please?

Best regards,

Renzo

lewisxhe, 2025-02-22:
7080G does not have PPS, and all LilyGo modems do not have PPS connected to ESP. So you need an external GPS module

7080G board carries PMU battery management chip, automatic charge and discharge management, but there is a disadvantage that if it is turned off, it cannot be turned on automatically and must be turned on manually.

SPI, I2C are the most basic, and all include these functions

RenzoFurlani, 2025-02-24:
Hello , Many thanks for your quick answer.

Please, could you advice me what is the best LilyGo to do a GPS based precise time syncronization and with SD card inside it? 

Best regards,

Renzo

lewisxhe, 2025-02-25:
There is no PPS function in the LTE series.
For PPS function and SD, you can consider T-BeamS3

RenzoFurlani, 2025-02-25:
Many thanks, 
Some other questions about T Beam series?
The "S3" mean that ESP32 is S3 model? 
I see some T-Beams boards and I like theses. But in anything of that boards -i have seen a SD card socket.
T-BEAMS3 is not in the main web page of LilyGo, it is very new?

lewisxhe, 2025-02-26:
1. Yes , ESP32S3
2. Yes ,it have sd socket
3. It can be found here https://lilygo.cc/products/t-beam-supreme?_pos=5&_sid=1682ba2a0&_ss=r

RenzoFurlani, 2025-02-26:
Many thanks for your answers.

All the T BEAM have SD cards and Ublox with PPS connected to ESP32?

IF you have more details , more than the information of T-BEAMs put in the web page, I would appriciate you a lot.

lewisxhe, 2025-02-27:
Only T-BeamS3 has SD and PPS, the ESP32 version does not.

RenzoFurlani, 2025-03-13:
Dear LilyGo sales support.
We are planning tu buy 20 units of some of your dev boards, and I have some changes in our requirements.
We , up now, do not need for GPS receiver and PPS signal. It could be done by a external module, may be ublox, connected to ESP32 by UART.
But we need a LilyGO board with Battery Managament chip, with SD or TF CARD, and to be programmable with Visual Code and IDF Libraries (I think this last issue es no a problem for any of your boards, but I am asking you).
We do not need both, any kind of LoRa or similar comunication and LCD display. There are more basic boards without these functionalities? 
Just only need a SPI and UART port for external comunication.
Best regards,
Renzo

lewisxhe, 2025-03-17:
This may be what you are looking for https://lilygo.cc/products/t7-s3

RenzoFurlani, 2025-03-17:
Lweisxhe, many thanks.
Your recomendation is very nice dev board, with a nice price too, but it dont have:
SD or TD Card support.
Power Management System, with input for 18650 lithium battery and solar panel.

This two charcteristics are only we need for a LilyGo board. I would repeat that we dont need LoRa, GPS, GSM or SIM support, and any kind of display. 
There are some other boards that fit these two items? 
Best regards,
Renzo

lewisxhe, 2025-03-18:
This board has battery management and 18650, but no SD card holder. You said that only T-Beam is suitable for you at present, but it has LoRa and GPS.
https://lilygo.cc/products/t-energy-s3
https://lilygo.cc/products/t-beam-softrf

RenzoFurlani, 2025-03-18:
Dear Lweisxhe, 
Many thanks... so in conclusion, there are not a board from LiLygo that adapt to all my requeriments, and some other electronics components like SD CARD, GPS receiver,  or Power Managment devices must be integrated from other electronics manufactures, at least that LiLyGO can sell us as a sepated component. 
We have to work in whitch of these devices should be integrated as external devices, and which other would be assemted in th LiLyGo board.
One of the last questions is about what is the IDE for progrramming this devices? We must program them ESP32 in Visual Code, using IDF Framework. This IDE and toolset is applied for all the LiLyGo ESP32 boards?
Best regards,
Renzo

lewisxhe, 2025-03-19:
ESP32 basically supports it, there is no question of whether the board supports it or not.

RenzoFurlani, 2025-03-20:
Many thanks, 

I will do a complete revision of ech of your products, in order to select the best option for us. 
Our priorities are SD or TF card and PMU or some power managament , both, on board wih ESP32. Then, explore avaliable pin out for connecting GPS with PPS and also ADC over SPI . 
Lora and TFT or similar display will not use for us, but if this are on board, we cannot do anything.

RenzoFurlani, 2025-03-26:
Dear Lweisxhe,
We are planning to afford and buy twenty LiLyGO T8 ESP S3 units, with a T-BAT for battery module. We can connect both devices? are they compatible with the same voltage and connectors? 
In other hand, the model T8 support both , WiFi and Bluetooth (LE). We are planning to do some monitoring evaluation using BLE or BT for near communications.
Last question is about SPI. In the T8, one SPI port are connected physically and logically to SD card. That is almost obvious. But what about the other SPI? Can we map it to some of physical pin and then connect ADC boards to it ? 
The T8 model are two version, and it depends if ESP 32 are S3 or C3. If it is S3, the NC pins disappear and they are now ready to use ? That is important for us, beacause we are planning to use several pins from ESP32 to connect external devices such as ADC boards, GPS, leds, and other things.
Many thanks.

lewisxhe, 2025-03-27:
ESP32S3 SPI can be freely mapped, and there are three available SPI masters that can be freely assigned.

T-BAT is just a power supply module, but it has an additional solar charging function. T8 also has its own charging function. If you want to use it together, it is also possible.

However, I suggest you first evaluate whether you need to use it this way

RenzoFurlani, 2025-03-28:
Hello, 
We are going to buy some of your boards.
Your products will be shipped to Argentina, and there are only one option in your webpage, by Fedex. 
Do you have other option than this? Fedex is not working well in Argentina, doing expensive custom local cost, and shipping cost is huge in comparasion to others companies. 
DHL is working wonderful for here . Could you work with this company? 
Many Thanks,

lewisxhe, 2025-04-23:
You need to contact the salesperson at the lilygo store, I am not responsible for these.

closed 2025-06-06 by github-actions[bot]