Xinyuan-LilyGO/LilyGo-T-SIM7080G #117: Deep sleep pwer constumption

ClemensGruber, 2025-02-23:
You can read at the T-SIM7080G's spec at https://lilygo.cc/products/t-sim7080-s3

> power consumption 
> - PSM 3.2 uA
> - sleep 600 uA

What means "PSM" power saving mode? and how can I obtain the 3.2 uA? 

What scetches are needed? This both one? 

- https://github.com/Xinyuan-LilyGO/LilyGo-T-SIM7080G/blob/master/examples/MinimalDeepSleepExample/MinimalDeepSleepExample.ino
- https://github.com/Xinyuan-LilyGO/LilyGo-T-SIM7080G/blob/master/examples/MinimalModemSleepMode/MinimalModemSleepMode.ino

Can I wake up from the PSM mode e.g. time based? So working / sending data > go to sleep in PSM / with 3.2 uA > wake up after 60 min > start over?

lewisxhe, 2025-02-24:
```
PSM 3.2 uA
sleep 600 uA
```

This information comes from the power consumption given by the SIM7080 official website, and does not represent the power consumption current of the entire board.

lewisxhe, 2025-02-24:
There is a discussion about sleep power consumption here https://github.com/Xinyuan-LilyGO/LilyGo-T-SIM7080G/issues/13

Regarding the problem of misleading current consumption in the mall, I have already reported it to LilyGo.

closed 2025-04-10 by github-actions[bot]