Xinyuan-LilyGO/LilyGo-T-SIM7080G #123: SD Card not working

carmi55, 2025-03-20:
Platform IO program: examples/MinimalSDCardExample

Seriel Monitor output:
>>E (16452) sdmmc_req: sdmmc_host_wait_for_event returned 0x107
E (16452) sdmmc_sd: sdmmc_check_scr: send_scr returned 0x107
E (16453) vfs_fat_sdmmc: sdmmc_card_init failed (0x107).
[ 16557][E][SD_MMC.cpp:138] begin(): Failed to initialize the card (0x107). Make sure SD card lines have pull-up resistors in place.
Card Mount Failed >>

Missing pull up resistors on TF-Card holder DAT3 (+DAT1 + DAT2) lines !
According to schematics the LILIGO T-SIM7080 TF card holder has only pull up resistors on SD_DAT0 (40),SD_CLK(38) and SD_CMD(39).

"SD pull-up requirements apply to cases where ESP32 uses the SPI or SDMMC controller to communicate with SD cards. When an SD card is operating in SPI mode or 1-bit SD mode, the CMD and DATA (DAT0 - DAT3) lines of the SD bus must be pulled up by 10 kOhm resistors. SD cards and SDIO devices should also have pull-ups on all above-mentioned lines (regardless of whether these lines are connected to the host) in order to prevent them from entering a wrong state."

https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/peripherals/sd_pullup_requirements.html

carmi55, 2025-03-20:
https://github.com/Xinyuan-LilyGO/T-Embed-CC1101/blob/master/hardware/T-Embed-CC1101%20V1.0%2024-07-29.pdf   has D3 connected to CS !

lewisxhe, 2025-03-22:
What SD card do you use? We tested using SanDisk SD and found no problems. You are talking about using SD 4-bit mode, each GPIO needs to be pulled up, while 1-bit mode only needs to set the 3 GPIOs used to be pulled up.

grf14, 2025-04-04:
Almost same problem for me. I'm testing the example code and this is the output: 
E (3098) sdmmc_sd: sdmmc_check_scr: send_scr returned 0x109
E (3099) vfs_fat_sdmmc: sdmmc_card_init failed (0x109).
[  3210][E][SD_MMC.cpp:138] begin(): Failed to initialize the card (0x109). Make sure SD card lines have pull-up resistors in place.
Card Mount Failed

grf14, 2025-04-07:
I made it work. I modify the frequency here:

SD_MMC.begin("/sdcard", true, false ,SDMMC_FREQ_PROBING)

lewisxhe, 2025-04-08:
So what frequency did you change to?

We tested using a SanDisk SD card, and the default frequency was able to pass the test.

grf14, 2025-04-08:
From 20MHz (default) to 400kHz. I follow this recommendation: https://github.com/espressif/esp-idf/issues/2478#issuecomment-424918316

These are the predefined variables of the library, they are in "sdmmc_types.h":

#define SDMMC_FREQ_DEFAULT      20000       /*!< SD/MMC Default speed (limited by clock divider) */
#define SDMMC_FREQ_HIGHSPEED    40000       /*!< SD High speed (limited by clock divider) */
#define SDMMC_FREQ_PROBING      400         /*!< SD/MMC probing speed */

carmi55, 2025-04-08:
I used:
- San Disk Ultra 16GB   -> didn't work
- San Disk Extreme 16GB -> didn't work
- Old San Disk 1GB  (12 years old) ... worked  ??

lewisxhe, 2025-04-09:
It seems to be a compatibility issue. We are also using San Disk Ultra 16GB.

grf14, 2025-04-09:
I'm using a San Disk Ultra 16GB too.

carmi55, 2025-04-30:
Got the SD Cards finally working:

- Step1:  set format_if_mount_failed = true  -> card is recognized, but no files/directories can be read or written 
- Step2: set #define BOARD_MAX_SDMMC_FREQ  SDMMC_FREQ_DEFAULT -> everything in the SD_Card Test is working.

closed 2025-06-14 by github-actions[bot]