Xinyuan-LilyGO/LilyGo-T-SIM7080G #127: module keeps dropping connection

yizhougcp, 2025-04-11:
hi there, 
I procured the LilyGo-T-SIM7080G recently.  and was able "upload" the sample code from here to the board.   However, no matter what program I uploaded, the board is disconnecting from the PC every few seconds (5 seconds?)   The customer service had been trying give me a firmware upload link but that link turned out to be "403 Forbidden"   I tried press both buttons (BOOT and RST) and then let go of the RST one and it did NOT work.    there is another saying on the internet:  "try pulling GPIO9 low, reset and program it" and it did not work either.    I am very close to give up and give a 1 star review on Aliexpress and their customer service sent me here.  please help! thank you

lewisxhe, 2025-04-14:
https://github.com/Xinyuan-LilyGO/LilyGo-T-SIM7080G/tree/master/firmware

[image]

The USB is disconnected every few seconds. You need to check if the program crashed, otherwise the USB should not be disconnected.

yizhougcp, 2025-04-15:
Thanks a lot Lewis.  refreshing the firmware fixed it.

closed 2025-04-15 by yizhougcp