Xinyuan-LilyGO/LilyGo-T-SIM7080G #144: Location fix does not work on SIMCOM fw 1951B16SIM7080, ok on 1951B08SIM7080

gusgonnet, 2025-05-23:
I got two Lyligo devices.
On one, location fix works fine.
On the other, it does not work.
Both units are using this code: [MinimalModemGPSExample.ino](https://github.com/Xinyuan-LilyGO/LilyGo-T-SIM7080G/blob/master/examples/MinimalModemGPSExample/MinimalModemGPSExample.ino).

The situation did not improve when :
- I replaced their GPS antennas
- I switched their GPS antennas
- Both units were placed outdoors for 12 hours

Looking at their modem SIMCOM firmware versions I get the following.

The working unit:
`
T-SIM7080G Firmware Version: 
AT+CGMR
Revision:1951B08SIM7080
`

The unit that does NOT work:
`
T-SIM7080G Firmware Version: 
AT+CGMR
Revision:1951B16SIM7080
`

Very weird that the fw version that works (08) is OLDER than the one that doesn't (16).

Are there any known issues with SIMCOM firmware level **1951B16SIM7080** in regards to location fix?

Logs:

`
AT+CGNSINF
AT+CGNSINF
AT+CGNSINF
AT+CGNSINF

`

lewisxhe, 2025-05-24:
Try updating to the latest B17

https://github.com/Xinyuan-LilyGO/LilyGo-T-SIM7080G/blob/master/docs/sim7080_update_firmware.md

gusgonnet, 2025-05-26:
thanks for the hint. Yes, location on B17 worked fine.

closed 2025-05-26 by gusgonnet